Community-PIO-CH32V/platform-ch32v #1: Add RISC-V toolchain pacakge

maxgerhardt, 2023-02-19:
PIO only has the SiFive variant but we want https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases to be same as the the MounRiverStudio at least.

Toolchain can still be upgraded from that.

maxgerhardt, 2023-02-19:
The packages have been submitted to the PIO Registry for review under the `community-ch32v` organization and are waiting for approval.

maxgerhardt, 2023-03-04:
A temporary work-around has been implemented in ce56da04f343377acdd305ae24fcd279e2ac0cc0

microcai, 2024-02-02:
any update on this?

how to use latest toolchain? c++23 is needed.

maxgerhardt, 2024-02-03:
Attempted in ab4dafc5eea660b38659dc52b1c4dbbe01e6265a, 83666d585f308ad25ef66d73acbffa3b89ea76e8 and 9b032a19b9595f6c08de0a18dd3bbd2f9bd9ae0f.

The update to GCC12 was a [massive failure](https://github.com/Community-PIO-CH32V/platform-ch32v/actions/runs/7769735424). The Windows toolchain works fine. The Linux toolchain misses the libprintf.a file that the OpenWCH Arduino core requires and thus fails building. The Mac toolchain fails catastrophically and can't even compile core_riscv.c. 

I've disabled the usage of GCC12 again in c4d8867cbaf22ab95bea08894bd7817530802afb after the fauled upgrade attempt. I've emailed support about these issues already and am awaiting another toolchain update, in, maybe a month, optimistically.

maxgerhardt, 2024-02-04:
>Hello, thank you for using and providing feedback. We will launch a new Linux community version and macOS, Linux toolchain package after the Spring Festival.
>
>MounRiver Studio

Okay, that sounds.. soon?

microcai, 2024-02-07:
does that means, if i use nonos-sdk, then gcc-12 is fine?

microcai, 2024-02-07:
> Attempted in [ab4dafc](https://github.com/Community-PIO-CH32V/platform-ch32v/commit/ab4dafc5eea660b38659dc52b1c4dbbe01e6265a), [83666d5](https://github.com/Community-PIO-CH32V/platform-ch32v/commit/83666d585f308ad25ef66d73acbffa3b89ea76e8) and [9b032a1](https://github.com/Community-PIO-CH32V/platform-ch32v/commit/9b032a19b9595f6c08de0a18dd3bbd2f9bd9ae0f).
> 
> The update to GCC12 was a [massive failure](https://github.com/Community-PIO-CH32V/platform-ch32v/actions/runs/7769735424). The Windows toolchain works fine. The Linux toolchain misses the libprintf.a file that the OpenWCH Arduino core requires and thus fails building. The Mac toolchain fails catastrophically and can't even compile core_riscv.c.
> 
> I've disabled the usage of GCC12 again in [c4d8867](https://github.com/Community-PIO-CH32V/platform-ch32v/commit/c4d8867cbaf22ab95bea08894bd7817530802afb) after the failed upgrade attempt. I've emailed support about these issues already and am awaiting another toolchain update, in, maybe a month, optimistically. This needs to work properly on all OSes, not only Windows. Until then, it's not done.
 
riscv-windows privode  **riscv-none-embed** while risc-linux-bin provide **riscv-none-elf**.

microcai, 2024-02-07:
copy  ```toolchain-riscv-windows/riscv-none-embed/lib/rv32imacxw/ilp32/libprintf.a``` to ```toolchain-riscv/riscv-none-elf/lib/rv32imac_xw/ilp32/``` and build success.

maxgerhardt, 2024-02-07:
I'll let MounRiver sort this one out so that we can do a clean update.

vihor3, 2024-02-21:
> 我会让 MounRiver 解决这个问题，以便我们可以进行干净的更新。

Currently, Linux and macos toolchains have been updated. I successfully compiled the firmware and ran it under macos using the toolchain of cmake and mousriver studio

microcai, 2024-03-05:
> I'll let MounRiver sort this one out so that we can do a clean update.

mainly because platform-build.py problem, force to add -lprintf flag。 and the result binary is way bigger than usual.

maxgerhardt, 2024-09-15:
The latest MacOS GCC12 toolchain is still broken. An Email has been sent to support.

maxgerhardt, 2025-06-30:
RISC-V GCC12 toolchain now works for all OSes and is the standard toolchain in the latest update.

AFter more verification with different boards, I will push GCC12 into the PIO registry.

microcai, 2025-07-09:
GCC14 is now the latest. still lag behind.  :(
why gcc 14 ? because  c++23 coroutine support is mainly broken below gcc 14.