Community-PIO-CH32V/platform-ch32v #12: Add documentation

maxgerhardt, 2023-02-19:
It should be documented what exact MCUs / eval / dev boards we support and what configuration options the framework each have.

Also how to get started with eval boards. Not setting the BOOT0/1 jumpers both to GND made me have some hickups in original bringup.

gicking, 2023-02-27:
just a note: AFAIK the CH32V00x devices don't have a built-in USB/UART bootloader, and have to be flashed via debugger.

gicking, 2023-02-27:
Question: do you know where I can find the bootloader protocol? Are you currently using the WCH tool? If yes, is that also available for Non-Win platforms?

maxgerhardt, 2023-02-27:
I'm trying to integrate the program `wchisp` (Open Source) for USB DFU uploads at #14, however it's not quite working for me yet. Not sure if also implements the UART upload.

Once that program has proven itself to at least work on my board, it'll be available cross-platform.

maxgerhardt, 2023-02-27:
I currently use the WCH-Link + OpenOCD combination to upload to my chip, which is already uploaded for most OS'es (#2), works great for upload and debugging.

maxgerhardt, 2023-04-01:
I've registered for a readthedocs.io website, now we have 

https://pio-ch32v.readthedocs.io/en/latest/

being created from https://github.com/Community-PIO-CH32V/docs. No real docs were added yet though, but the page stands.

maxgerhardt, 2024-03-10:
Needs docs for "How to convert MounRiverStudio project to PlatformIO" desperately, see #53.

FPGA-Computer, 2024-09-04:
> Sorry, wrong information: the CH32V003 DS v1.2 states on p.1 that it does feature a bootloader. However, no further information is provided

The chip doesn't have a Boot pin to select Bootloader externally. Instead there is a non-volatile Mode bit that can be set to either jump to user code or bootload upon Reset.  (16.3.4 OBKEY register)

See SystemReset_StartMode(uint32_t Mode) under  /EXAM/SRC/Peripheral/src/ch32v00x_flash.c in the EVT application code example.