Community-PIO-CH32V/platform-ch32v #13: Add Arduino support

maxgerhardt, 2023-02-19:
Once all existing frameworks are supposed, one can think about expanding  / creating an Arduino core for these boards.

maxgerhardt, 2023-05-21:
Done for CH32V003 in https://github.com/Community-PIO-CH32V/platform-ch32v/commit/2f95c188d471f2e1087c3d5803c5c2345387887a

marshalab, 2023-06-29:
Hello. When can we expect the addition of functions analogRead analogWrite AttachInterrupt for Arduino part ? I've tried changing the adc-tempo example to read voltages, but it's hard for me. Perhaps there are working pieces of code for these functions?

maxgerhardt, 2023-06-29:
https://github.com/AlexanderMandera/arduino-wch32v003 doesn't have `analogRead()` yet, so if you want it in that core, you need to open an issue there.

I don't yet support building https://github.com/openwch/arduino_core_ch32 but it will be supported soon. This core **does** have analogRead(). You can use this core in the Arduino IDE already.

https://github.com/openwch/arduino_core_ch32/blob/987154335fd6834d2f547e559376d553c9356a1b/cores/arduino/wiring_analog.c#L123C1-L136

marshalab, 2023-06-29:
Great, thanks a lot. 
https://github.com/openwch/arduino_core_ch32
This is what I need now!

arwidcool, 2023-10-17:
https://github.com/openwch/arduino_core_ch32  Now has native Arduino SPI implementation..

I tried adding it as a custom core to CH32V framework but failed. I think adding it now, especially for CH32V003 will be a big step in getting low cost projects going.

I can try adding it myself if there is any documentations on doing so, if not I think it will be very beneficial to add this core at least for V003.

maxgerhardt, 2023-10-17:
This core is *already* supported for CH32V003. Please see project's https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/examples/blinky-arduino platformio.ini file.

maxgerhardt, 2024-09-15:
Closing since a lot of boards are supported in the OpenWCH Arduino core now. New boards will be synced here as they gain support in that core.

closed 2024-09-15 by maxgerhardt