Community-PIO-CH32V/platform-ch32v #16: Broken links in Readme.md

gicking, 2023-02-26:
The following links in Readme.md don't exist
  - [Home](https://registry.platformio.org/platforms/platformio/ch32v) (home page in the PlatformIO Registry)
  - [Documentation](https://docs.platformio.org/page/platforms/ch32v.html) (advanced usage, packages, boards, frameworks, etc.)
  - [documentation](https://docs.platformio.org/page/platforms/ch32v.html) at bottom of page

I suspect(!) these will be created once ch32v platform is released...? If yes, please just close this issue

maxgerhardt, 2023-02-26:
Right. I just copy-pasted the template from the other platforms and replaced the name.

This definitely needs a correct / custom README as I think integration of this platform into the official PlatformIO repos will be here soon.

Tracked in #12 too.

maxgerhardt, 2023-03-04:
Fixed through 06ca19ef8f231521c78374d65129a8449aaa5d14

closed 2023-03-04 by maxgerhardt