Community-PIO-CH32V/platform-ch32v #17: General description in Readme.md

gicking, 2023-02-26:
The family description at the top of the page is somewhat mixed up:
_CH32V series are industrial-grade general-purpose microcontrollers designed based on QingKe 32-bit RISC-V. The whole series of products into the hardware stack area, fast interrupt entry and other designs, compared to the standard greatly improved the interrupt response speed._

Proposal:
_CH32V series are industrial-grade, general-purpose microcontrollers based on a range of QingKe 32-bit RISC-V cores. All devices feature a hardware stack area and fast interrupt entry, which greatly improves interrupt latency. The family ranges from ultra-cheap, low-end CH32V00x with 2kB RAM / 16kB flash, up to high speed, ultra-connected CH32V30x with 64kB RAM / 256kB flash, hardware FPU, USB, CAN, Ethernet, etc._
_Note: WCH also offers the CH32F family, which is ARM based with identical peripherals_

maxgerhardt, 2023-02-26:
Can we get the V103 and V20x (V203, V208) in the text too somehow? Then it'd be perfect.

Btw I copied the description from the V303 datasheet

[image]

gicking, 2023-02-27:
I'll write up something - it's the least I can do to support  ;-)

Actually the range of speeds and features is so wide between CH32V00x and CH32V30x, that I'll write a short paragraph for each one.

maxgerhardt, 2023-02-27:
That info is also in [the chip database](https://github.com/Community-PIO-CH32V/platform-ch32v/blob/f1eb50cee3b0a9a068fadc87f54f8c89e3cab310/misc/scripts/gen_boarddefs.py#L65-L103) I copied by hand from the datasheets, maybe it helps.

gicking, 2023-02-27:
it certainly would have - if only I had read your comment earlier...  ;-)

HOWEVER, I find some conflicting information :-(   For example, the CH32V203 [Github summary](https://github.com/openwch/ch32v20x) states "20KB RAM, 64KB Flash", whereas the [Datasheet](http://www.wch-ic.com/downloads/CH32V203DS0_PDF.html) specifies "up to 64KB RAM, 224KB Flash". Any idea on that?

In any case, I was thinking about a tabular overview like attached file (data taken from datasheets)
[Readme.md](https://github.com/Community-PIO-CH32V/platform-ch32v/files/10838386/Readme.md)

maxgerhardt, 2023-02-27:
Hm

[image]

[image]

gicking, 2023-02-27:
The marked text implies that you could re-configure RAM as Flash...? Weird!

Besides, it still doesn't match the text on page 1 of the DS

maxgerhardt, 2023-02-27:
Yes, reconfigurable memory sizes is something I find weird too (especially its technical implementation -- RAM can't just become persistant Flash with a flick of a switch). But it's definitely the case and documented in the linker scripts too, see #20.

gicking, 2023-02-28:
if it isn't PHYSICAL memory - which BTW would be mind blowing - I guess it must be an overlap of LOGICAL addresses. However, that would be some restriction in the internal design, because the CH32V203 memory map allows for >380MByte of code flash (see below). And even then it wouldn't overlap with RAM range. Very strange... 

[image]

Maybe it's something I should ask WCH support...

gicking, 2023-03-02:
initial reply from tech support:
_Among CH32V20X series chips, only CH32V203RBT6 can configure flash and RAM sizes. Users configure the Flash and RAM sizes according to their needs. The space can be used more reasonably and the program running speed can be improved._

reply to follow-up question: 
_this is the internal design mechanism of the chip, and the specific principle is a design secret. There are only ways in the data sheet to configure the size of Flash and RAM. I believe you have found that the sum of the configured Flash and RAM sizes is fixed. Unreasonable allocation may affect your use._

That doesn't really answer or question...   :-/

gicking, 2023-03-04:
My **hypothesis** for the described behaviour is that CH32V doesn't directly use physical address but rather a mapping between a logical and physical address similar to the D-flash of the TLE98xx family. And, for reasons beyond my understanding, the logical address range in the affected device is limited to Flash+RAM = 192kB. It's the only reason I can think of...

Of course that is, unless WCH uses an advanced memory technology where volatile and non-volatile memory are similar or equal, e.g. FRAM or MRAM. But I guess they'd be marketing such a superior feature openly. 

Any other ideas? Or an idea how to test the hypothesis?

maxgerhardt, 2023-03-04:
I don't think we need to get into the knitty-gritty details of understanding this curiosity of reconfigurable memory sizes, it's a trade secret after all. We just need the description to accurately reflect the available devices. The datasheet's device listings should be the source of truth here.

gicking, 2023-03-07:
Just one more comment: CH32V is definitely using memory (re-)mapping. Specifically bootloader flash can be mapped to the reset vector address based on BOOTx pins (CH32V1xx/2xx/3xx) or flash flag (CH32V0xx), see the respective datasheets.

However, why they restricted the CH32V203RBT6 memory to use either RAM or flash is beyond me...

gicking, 2023-03-10:
Hi again,

coming back to the original topic of this issue: below please find an updated version of the `Readme.md`.

I found contradictions within the datasheets, e.g. the V30x DS states flash/RAM up to  480KB/128kB at several locations. However, the (currently) available devices "only" range up to 256kB/64kB. So either it's a typo, or WCH plans bigger devices in the future.

Anyway, because of that I omitted the device overview list, but just link to the WCH product selector page. 

[README.md](https://github.com/Community-PIO-CH32V/platform-ch32v/files/10939472/README.md)

maxgerhardt, 2023-03-10:
Thanks! Description updated in https://github.com/Community-PIO-CH32V/platform-ch32v/commit/e625d3445826f8be44488f6f0c7d9795632cb0e0.

closed 2023-03-10 by maxgerhardt