Community-PIO-CH32V/platform-ch32v #19: Support RISC-V FPU (Hard-Float)

maxgerhardt, 2023-02-27:
The CH32V30X QingKe V4F cores are capable of (single-precision) floating point in hardrware, yet the original SDK only ever configures them for software.

See 

https://github.com/Community-PIO-CH32V/platform-ch32v/blob/f1eb50cee3b0a9a068fadc87f54f8c89e3cab310/misc/scripts/gen_boarddefs.py#L39-L47

We should have a build option to switch to hardfloat.

tmolteno, 2023-04-02:
Good to know about this issue.  I've been reading a bit (https://www.sifive.com/blog/all-aboard-part-1-compiler-args) and there is an example where:

```
-march=rv32imafdc -mabi=ilp32: Hardware floating-point instructions can be
 generated, but no floating-point arguments will be passed in registers. 
This is like the -mfloat-abi=softfp argument to ARM's GCC, and is usually 
used when interfacing with soft-float binaries on a hard-float system.
```
If get_riscv_arch_and_abi() returned the correct arch as "rv32imafcxw", and kept ilp32, would this "Just work" as in the example above where only the code we wrote would get hard float code?

maxgerhardt, 2023-04-02:
It would definitely seem so to me.

As a sidenote, I did see multiple W.CH provided precompiled libraries also come in a `_float` form, like https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/examples/webserver-ch32v307-none-os/lib/NetLib, so even switching to a `hardfp` ABI would still work and might result in additional performance benefits. Still have to test everything out though.

And the path I'd go here would not be to change `get_riscv_arch_and_abi()` (which generates the board JSON's default values) but have a configuration option via the `platformio.ini` to activate hardfloat instructions and/or ABI, like
```
board_build.use_hardfloat_arch_and_abi = Yes
```

maxgerhardt, 2023-05-21:
```ini
; activate hardfloat instructions (march)
; and passing floating point args via fp registers (mabi)
board_build.march = rv32imafcxw
board_build.mabi = ilp32f
```
works fine and generates floating point instructions, I will just document this.

maxgerhardt, 2023-05-21:
Documented in https://github.com/Community-PIO-CH32V/docs/commit/7f03e13d472b76cfd1b3a1ef662deccb61502445

closed 2023-05-21 by maxgerhardt