Community-PIO-CH32V/platform-ch32v #24: Linux debug failed

gicking, 2023-04-05:
hello,

thanks to [issue #22](https://github.com/Community-PIO-CH32V/platform-ch32v/issues/22), upload to CH32V003 now works also under Linux (thanks a lot!). Note also [this comment](https://github.com/Community-PIO-CH32V/platform-ch32v/issues/22#issuecomment-1493454979) on `udev` rules 

However, if I try to start a debug session, I get an error message 

`Error: Could not launch Debug Server 'bin/openocd'. Please check that it is installed and is included in a system PATH` 

The only executable `openocd`  I found under `~/.platformio` is in `~/.platformiopackages/tool-openocd/bin`, so I added this to `PATH` via `.profile`. Still, same error message  :-(

As you can see in the screenshot, `openocd` path is indeed found. But PIO seemingly is not aware of it, even after a reboot. 

[image]

maxgerhardt, 2023-04-05:
PlatformIO will only use the OpenOCD version of the installed PlatformIO package, never a system-wide installed tool.

But I definitely see that I have a bug here, on a fresh Linux installation it's trying to clone it from the repo again that is for Windows and then it doesn't find it. Since it's in the PIO registry now the fix should be easy.

```
undefinedTool Manager: Installing git+https://github.com/Community-PIO-CH32V/tool-openocd-riscv-wch.git
git version 2.37.2
undefinedKlone nach '/home/max/.platformio/.cache/tmp/pkg-installing-qcxw7b70' …
Error: Could not launch Debug Server 'bin/openocd'. Please check that it is installed and is included in a system PATH
See https://docs.platformio.org/page/plus/debugging.html
```

maxgerhardt, 2023-04-05:
Yes the latest commit f72fb36e724ea23ee5bb05f298f6133c61414f98 has fixed this.

[image]

You can try updating with a [PlatformIO Core CLI](https://github.com/Community-PIO-CH32V/ch32-pio-projects#update-via-platformio-core-cli) -> `pio pkg update -g -p ch32v`

gicking, 2023-04-05:
debugging works after updating. Thanks a lot!  :-)

closed 2023-04-05 by gicking