Community-PIO-CH32V/platform-ch32v #25: "can't link RVE with other target" when building for CH32V003A4M6 on mac. blinky-none-os example.

symonty, 2023-05-05:
Otehr boards seem to compile fine. the riscv-non-embed binaries seem to be installed correctly.

At a a loss why this will not link, I am missing something or a bug? 

`.platformio/packages/toolchain-riscv@src-4a976c8edcf1083542a06c72f741fdf2/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld: .platformio/packages/toolchain-riscv@src-4a976c8edcf1083542a06c72f741fdf2/bin/../lib/gcc/riscv-none-embed/8.2.0/libgcc.a(save-restore.o): can't link RVE with other target`

same error for 

.. muldi3.o ... div.o ... lib_a-init.o

maxgerhardt, 2023-05-05:
I've observed this failure too in the CI but only with some examples. Something is definitely wrong there with the used libgcc.a file, for Mac we might need to substitute the one from the Linux Toolchain (which should be fine).I don't have a Mac to test so I'd love for you to do it: Can you go to the official mounriver website, download the IDE for Mac, create s new CH32V003 project (from their new.project template), disable Build->Concise Output, then build and send the log here (via Pastebin.com)?

symonty, 2023-05-06:
`official mounriver website, download the IDE for Mac,`

Sorry I dont think there is an [official IDE,](http://www.mounriver.com/download) however they do offer the [toolchain as xPack](
http://file.mounriver.com/tools/MRS_Toolchain_MAC_X64_V170.zip) which looks to be the same as the toolchain i got from the platformio install. 

~/.platformio/packages/toolchain-riscv@src-4a976c8edcf1083542a06c72f741fdf2

I can try replace the toolchain with the one downloaded... I will report back...

_ofcourse it could be ( but unlikely ) a rosetta issue since the toolchains are X64 and M1/M2 are ARM?_

unixbigot, 2023-05-06:
I get the same issue on my intel mac, so Rosetta is off the hook.

symonty, 2023-05-06:
> I get the same issue on my intel mac, so Rosetta is off the hook.

OK thx for that working on getting it running on UTM on my M1 to confirm it aint my project, now I know it aint a ARM vs X86 issue.

unixbigot, 2023-05-06:
I've installed the mounriver xpack and tried it out.   No IDE (just toolchain) so I grabbed an example command line-friendly project from https://github.com/cjacker/ch32v_evt_makefile_gcc_project_template

This project's setup script does not run on Mac, so I debugged it, making these changes: https://github.com/unixbigot/ch32v_evt_makefile_gcc_project_template/commit/dbcfc5f70f2116741cb8bd091cf66835cbb4c3b9

With all that, building a skeleton app runs these commands:

```
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sect
ions -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/core_riscv.d"  -Wa,-a,-ad,-alms=build/core_riscv.lst CH32V_firmware_library/Core/core_riscv.c -o build/core_riscv.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sect
ions -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ch32v00x_dma.d"  -Wa,-a,-ad,-alms=build/ch32v00x_dma.lst CH32V_firmware_library/Peripheral/src/ch32v00x_dma.c -o build/ch32v00x_dma.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sect
ions -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ch32v00x_misc.d"  -Wa,-a,-ad,-alms=build/ch32v00x_misc.lst CH32V_firmware_library/Peripheral/src/ch32v00x_misc.c -o build/ch32v00x_misc.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sect
ions -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ch32v00x_adc.d"  -Wa,-a,-ad,-alms=build/ch32v00x_adc.lst CH32V_firmware_library/Peripheral/src/ch32v00x_adc.c -o build/ch32v00x_adc.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sect
ions -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ch32v00x_pwr.d"  -Wa,-a,-ad,-alms=build/ch32v00x_pwr.lst CH32V_firmware_library/Peripheral/src/ch32v00x_pwr.c -o build/ch32v00x_pwr.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sect
ions -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ch32v00x_iwdg.d"  -Wa,-a,-ad,-alms=build/ch32v00x_iwdg.lst CH32V_firmware_library/Peripheral/src/ch32v00x_iwdg.c -o build/ch32v00x_iwdg.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sect
ions -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ch32v00x_wwdg.d"  -Wa,-a,-ad,-alms=build/ch32v00x_wwdg.lst CH32V_firmware_library/Peripheral/src/ch32v00x_wwdg.c -o build/ch32v00x_wwdg.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sect
ions -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ch32v00x_exti.d"  -Wa,-a,-ad,-alms=build/ch32v00x_exti.lst CH32V_firmware_library/Peripheral/src/ch32v00x_exti.c -o build/ch32v00x_exti.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sect
ions -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ch32v00x_flash.d"  -Wa,-a,-ad,-alms=build/ch32v00x_flash.lst CH32V_firmware_library/Peripheral/src/ch32v00x_flash.c -o build/ch32v00x_flash.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ch32v00x_dbgmcu.d"  -Wa,-a,-ad,-alms=build/ch32v00x_dbgmcu.lst CH32V_firmware_library/Peripheral/src/ch32v00x_dbgmcu.c -o build/ch32v00x_dbgmcu.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ch32v00x_spi.d"  -Wa,-a,-ad,-alms=build/ch32v00x_spi.lst CH32V_firmware_library/Peripheral/src/ch32v00x_spi.c -o build/ch32v00x_spi.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ch32v00x_tim.d"  -Wa,-a,-ad,-alms=build/ch32v00x_tim.lst CH32V_firmware_library/Peripheral/src/ch32v00x_tim.c -o build/ch32v00x_tim.o
riscv-none-embed-gcc -c -march=rv32ec -mabi=ilp32e -msmall-data-limit=8    -ICH32V_firmware_library/Peripheral/inc -ICH32V_firmware_library/Debug -ICH32V_firmware_library/Core -IUser -Os -Wall -fdata-sect
[... cut ...]
-sections -Wunused -Wuninitialized -T CH32V_firmware_library/Ld/Link.ld  -nostartfiles -Xlinker --gc-sections -Wl,-Map=build/ch32v003f4u6.map --specs=nano.specs -lc -lm -lnosys -o build/ch32v003f4u6.elf
riscv-none-embed-size build/ch32v003f4u6.elf
   text    data     bss     dec     hex filename
   9652     152     276   10080    2760 build/ch32v003f4u6.elf
riscv-none-embed-objcopy -O ihex build/ch32v003f4u6.elf build/ch32v003f4u6.hex
riscv-none-embed-objcopy -O binary -S build/ch32v003f4u6.elf build/ch32v003f4u6.bin
```

unixbigot, 2023-05-06:
Comparing the link stage arguments, pio is using `-march=rv32ecxw` while mounriver is using `-march=rv32ec`.

Amending the pio link command to use the same argument makes it appear to succeed.

unixbigot, 2023-05-06:
Following that back to the board file, the following change (applied in ~/platformio/platforms/ch32v) gives me a successful `pio run -t upload`, and code that runs on the device:

diff --git a/boards/genericCH32V003F4U6.json b/boards/genericCH32V003F4U6.json
index 4bba765..98d915c 100644
--- a/boards/genericCH32V003F4U6.json
+++ b/boards/genericCH32V003F4U6.json
@@ -9,7 +9,7 @@
       ]
     ],
     "mabi": "ilp32e",
-    "march": "rv32ecxw",
+    "march": "rv32ec",
     "mcu": "ch32v003f4u6",
     "series": "ch32v003"
   },

symonty, 2023-05-06:
Confirmed my code for the A4M6 now compiles. 

Will try and upload tomorrow ( PST time here, not AEST ) Thx.

manually removed xw from  march in ./platforms/ch32v/boards/genericCH32V003A4M6.json

maxgerhardt, 2023-05-06:
That's... but the MounRiver IDE uses this exact architecture. The `xw` architecture part is for the custom extensions that the CH32V003 has.

[image]

[QingKeV2_Processor_Manual.PDF](https://github.com/Community-PIO-CH32V/platform-ch32v/files/11412482/QingKeV2_Processor_Manual.PDF)

So when we're not turning that on, we're losing in terms of firmware size and probably execution speed (shorter instructions are faster to load).

I can dynamically turn off `xw` for Mac in the builder script and probbaly make it go through, but it doesn't feel right at all. I'll need to ask MounRiver support again what's going on there.

maxgerhardt, 2023-05-06:
100% what MounRiver IDE is using.

[image]

maxgerhardt, 2023-05-07:
Fixed per PR above until I get a response from MounRiver support or a fixed MacOS toolchain that can do `rv32ecxw`.

closed 2023-05-07 by maxgerhardt

maxgerhardt, 2023-05-17:
[image]

New toolchain dropped, this will likely fix the problem of missing rv32ecxw libraries -- will look into updating the toolchain, re-upgrading the `march` and retesting in CI.

maxgerhardt, 2023-05-21:
Toolchain updated per https://github.com/Community-PIO-CH32V/toolchain-riscv-mac/commit/99cb62f72304e1a7afe8e7c191143975793ccfab and workaround removed per https://github.com/Community-PIO-CH32V/platform-ch32v/commit/0f2a3e14d62e9f19077e3d01ea5664e0bc2ebbed.

Examples  [still compile](https://github.com/Community-PIO-CH32V/platform-ch32v/actions/runs/5037142983/jobs/9033781372) normally, so everything is fixed now properly :)

unixbigot, 2023-05-24:
Works for me.    

For those reading along, I reverted my platform hack, then did `pio pkg update -g -p https://github.com/Community-PIO-CH32V/platform-ch32v.git` which updated some stuff.   Compilation still failed---looks like the toolchain didn't update.      I deleted `.platformio/packages/toolchain-riscv/` then recompiled my project, and it fetched the new toolchain during compilation and compiles OK now.

PR #26 (2023-05-07): Fix CH32V003 compilation on Mac
..by downgrading the architecture to `rv32ec` instead of the full `rv32ecxw`.

Ugly, but temporary workaround until I get a message back about this from MounRiver support.

Fixes #25.