Community-PIO-CH32V/platform-ch32v #28: minichlink debug support

maxgerhardt, 2023-06-09:
With https://github.com/Community-PIO-CH32V/platform-ch32v/commit/d451c465657bf796c1f6b57f0f0b710ab09e6d4c, we can now upload via minichlink. We can also start minichlink's GDB server and connect to it, however, the step into and step over controls in VSCode are broken because minichlink just hangs on the GDB `step` command. Thus, we depend on https://github.com/cnlohr/ch32v003fun/issues/158.

Other oddities like not supporting `load` were worked-around with the `preload` load mode in PlatformIO.

It's also an open question on how to do a reset as GDB command.

tmolteno, 2023-06-15:
With the most recent version I am getting errors on trying to upload (I had been using the using Link-E) On debian linux. 

`warning: Could not find remote branch master to clone.
fatal: Remote branch master not found in upstream origin
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'master', 'https://github.com/Community-PIO-CH32V/tool-minichlink.git', '/home/tim/.platformio/.cache/tmp/pkg-installing-r9gmgsmc']

Seems a bit random when this is happening... Choosing "Run without debugging" seems to still upload. Hitting the upload button causes this error.

`

tmolteno, 2023-06-15:
I think the problem is that the branch is called "main". Modifying this in the line 

```
platforms/ch32v/platform.json:      "version": "https://github.com/Community-PIO-CH32V/tool-minichlink.git#master",
```
to 
```
platforms/ch32v/platform.json:      "version": "https://github.com/Community-PIO-CH32V/tool-minichlink.git#main",
```

Allows the upload to proceed further, however all are now failing. I need to check my debugger using the MounRiver tools...

maxgerhardt, 2023-06-15:
Thanks for the info, I'll investigate this later.

maxgerhardt, 2023-06-15:
Might already be fixed with 131d250034d057222b27b8b75cb557290e333845, can you reinstall the platform cleanly? 
```
rm -rf ~/.platformio/platforms/ch32v
rm -rf ~/.platformio/packages/tool-minichlink
```

maxgerhardt, 2024-11-15:
Debugging inside Windows with minichlink partly working with https://github.com/cnlohr/ch32v003fun/pull/444, `load` GDB command still broken.