Community-PIO-CH32V/platform-ch32v #30: Problems with uploading firmware ch32v003f4p6

Alex-IC, 2023-06-18:
### Discussed in https://github.com/Community-PIO-CH32V/platform-ch32v/discussions/29

<div type='discussions-op-text'>

<sup>Originally posted by **Alex-IC** June 18, 2023</sup>
I’m trying to upload code via Platformio ->Upload and getting an error:

```
Total         1875
<lambda>(["upload"], [".pio\build\genericCH32V003F4P6\firmware.elf"])
AVAILABLE: isp, minichlink, wch-link
CURRENT: upload_protocol = wch-link
openocd -c "debug_level 2" -s C:\Users\Alexey\.platformio\packages\tool-openocd-riscv-wch\bin -s C:\Users\Alexey\.platformio\packages\tool-openocd-riscv-wch\scripts -f wch-riscv.cfg -c init -c halt -c "program {.pio\build\genericCH32V003F4P6\firmware.elf} verify reset" -c shutdown
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-01-03-10:00)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Info : WCH-LinkE  mode:RV version 2.7 
Info : wlink_init ok

*** [upload] Error 1
========================================================================== [FAILED] Took 2.58 seconds ======================

```
What could be the problem? Through the WCH Link Utility it is stitched without problems.
Chip ch32f008f4p6
VSCode 1.79.2
PlatformIO 6.1.7
OS Windows_NT x64 10.0.0.17763</div>

maxgerhardt, 2023-06-18:
>Info : WCH-LinkE  mode:RV version 2.7 

The firmware on mine is 2.8.

```
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Info : WCH-LinkE  mode:RV version 2.8
```
I think since we're using the latest OpenOCD you also need the latest firmware version. 

Can you install the very latest version of MounRiver Studio Version and check if you can do a firmware update of your WCH-LinkE to v2.8?

maxgerhardt, 2023-06-18:
Huh, even I am not on the latest version. After the latest MounRiver Studio update, when I go Flash->Download in a CH32V003 template project, it asks me to update too.

[image]

[image]

Welp, let's try this out then.. Worst case I have to update OpenOCD again everywhere.

maxgerhardt, 2023-06-18:
And PlatformIO still works fine with the updated firmware version.

[image]

closed 2023-06-19 by Alex-IC