Community-PIO-CH32V/platform-ch32v #31: Unable to flash firmware on MacOS

IndianTinker, 2023-06-26:
Hello all,
Thanks for the amazing work. I am using platformIO with VSCode on a M2 Pro MacBook Pro.
I have got the LED blinking using the Arduino IDE using [this](https://github.com/AlexanderMandera/arduino-wch32v003 ) port of the Arduino core. The [official core](https://github.com/openwch/arduino_core_ch32) does not seem to support MacOS yet.

Hence, there is no issue on the physical level. I followed all the steps in the README and in addition I followed [this](https://blog.csdn.net/weixin_54203946/article/details/129844078?utm_medium=distribute.pc_relevant.none-task-blog-2~default~baidujs_utm_term~default-0-129844078-blog-128550404.235^v38^pc_relevant_sort_base2&spm=1001.2101.3001.4242.1&utm_relevant_index=3) excellent resource to adapt the guide to MacOS.  

On VSCode, when I compile, it works perfectly. But when I hit flash/upload, I get this error :
`erbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ch32v/ch32v003f4p6_evt_r0.html
PLATFORM: WCH CH32V (1.0.0+sha.131d250) > CH32V003F4P6-EVT-R0
HARDWARE: CH32V003F4P6 48MHz, 2KB RAM, 16KB Flash
DEBUG: Current (wch-link) On-board (wch-link) External (minichlink)
PACKAGES: 
 - tool-minichlink @ 0.1.0+sha.af02ba5 
 - tool-openocd-riscv-wch @ 2.1100.230329 (11.0) 
 - tool-wchisp @ 0.22.230228 
 - toolchain-riscv @ 1.80200.190731+sha.99cb62f
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/ch32v003f4p6_evt_r0/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.5% (used 256 bytes from 2048 bytes)
Flash: [          ]   3.1% (used 516 bytes from 16384 bytes)
Configuring upload protocol...
AVAILABLE: isp, minichlink, wch-link
CURRENT: upload_protocol = wch-link
Uploading .pio/build/ch32v003f4p6_evt_r0/firmware.elf
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-03-02-14:04)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

Warn : Transport "sdi" was already selected
Ready for Remote Connections

*** [upload] Error 1`

If I press the button the CH32V003 board, I get the following error : 
`Processing ch32v003f4p6_evt_r0 (board: ch32v003f4p6_evt_r0; platform: ch32v)
-----------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ch32v/ch32v003f4p6_evt_r0.html
PLATFORM: WCH CH32V (1.0.0+sha.131d250) > CH32V003F4P6-EVT-R0
HARDWARE: CH32V003F4P6 48MHz, 2KB RAM, 16KB Flash
DEBUG: Current (wch-link) On-board (wch-link) External (minichlink)
PACKAGES: 
 - tool-minichlink @ 0.1.0+sha.af02ba5 
 - tool-openocd-riscv-wch @ 2.1100.230329 (11.0) 
 - tool-wchisp @ 0.22.230228 
 - toolchain-riscv @ 1.80200.190731+sha.99cb62f
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/ch32v003f4p6_evt_r0/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.5% (used 256 bytes from 2048 bytes)
Flash: [          ]   3.1% (used 516 bytes from 16384 bytes)
Configuring upload protocol...
AVAILABLE: isp, minichlink, wch-link
CURRENT: upload_protocol = wch-link
Uploading .pio/build/ch32v003f4p6_evt_r0/firmware.elf
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-03-02-14:04)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

Warn : Transport "sdi" was already selected
Ready for Remote Connections
Error:  WCH-Link failed to connect with riscvchip
Error:  1.Make sure the two-line debug interface has been opened. If not, set board to boot mode then use ISP tool to open it
Error:  2.Please check your physical link connection

*** [upload] Error 1`

What could be the possible error? Thanks a lot again! Cheers.

maxgerhardt, 2023-06-26:
Alexander's Arduino core uses minichlink instead of OpenOCD for upload, which might have different behavior. Can you quickly add

```
upload_protocol = minichlink
```
in your `platformio.ini` and reupload?

maxgerhardt, 2023-06-26:
> Hence, there is no issue on the physical level. I followed all the steps in the README and in addition I followed [...]

Also thanks for the blog post link, it seems people have to work around a few issues with my Platform. That's not supposed to be how it goes. 

At least the "Can't install tool-openocd-riscv-wch" problem should be **gone** because the package is in the registry for MacOS and all other platforms (https://registry.platformio.org/tools/community-ch32v/tool-openocd-riscv-wch/compatibility). Installation of HIDAPI and LIBUSB is something that should be better documented though.

IndianTinker, 2023-06-26:
Hey,
Thanks for a quick reply. 
I tried changing the upload method to minichlink and it failed too. 
The logs are as follows :
```

Processing genericCH32V003A4M6 (board: genericCH32V003A4M6; platform: ch32v; framework: arduino)
-----------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ch32v/genericCH32V003A4M6.html
PLATFORM: WCH CH32V (1.0.0+sha.131d250) > Generic CH32V003A4M6
HARDWARE: CH32V003A4M6 48MHz, 2KB RAM, 16KB Flash
DEBUG: Current (wch-link) On-board (wch-link) External (minichlink)
PACKAGES: 
 - framework-arduinoch32v003 @ 0.0.0+sha.cb439a8 
 - tool-minichlink @ 0.1.0+sha.634981a 
 - tool-openocd-riscv-wch @ 2.1100.230329 (11.0) 
 - tool-wchisp @ 0.22.230228 
 - toolchain-riscv @ 1.80200.190731+sha.99cb62f
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/genericCH32V003A4M6/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.4% (used 376 bytes from 2048 bytes)
Flash: [=         ]   8.7% (used 1420 bytes from 16384 bytes)
Configuring upload protocol...
AVAILABLE: isp, minichlink, wch-link
CURRENT: upload_protocol = minichlink
Uploading .pio/build/genericCH32V003A4M6/firmware.bin
/Users/rgupta/.platformio/packages/tool-minichlink/minichlink: line 1: 3: command not found
/Users/rgupta/.platformio/packages/tool-minichlink/minichlink: line 2: 0: command not found
/Users/rgupta/.platformio/packages/tool-minichlink/minichlink: line 3: 69632: command not found
/Users/rgupta/.platformio/packages/tool-minichlink/minichlink: line 4: 81920: command not found
/Users/rgupta/.platformio/packages/tool-minichlink/minichlink: line 5: 4096: command not found
/Users/rgupta/.platformio/packages/tool-minichlink/minichlink: line 6: 98304: command not found
/Users/rgupta/.platformio/packages/tool-minichlink/minichlink: line 7: 15216: command not found
/Users/rgupta/.platformio/packages/tool-minichlink/minichlink: line 8: syntax error near unexpected token `('
/Users/rgupta/.platformio/packages/tool-minichlink/minichlink: line 8: `\xcf\xfa\xed\xfe\xa0\x85 H__PAGEZERO(__TEXT__text__TEXT [\xc5 \x80__stubs__TEXT|\xdeF|\xd\x80__stub_helper__TEXT\xc4\xe0\xda\xc4\xe0\x80__cstring__TEXT\x9e\xe4\xa8\x9e\xe4__const__TEXTH\xffdH\xff__unwind_info__TEXT\xac\xffT\xac\xff\xe8__DATA_CONST@@__got__DATA_CONST8a__cfstring__DATA_CONST8 8\x88__DATA@@@@__la_symbol_ptr__DATA@h__data__DATC__common__DATA C__bss__DATA0I$H__LINKEDIT\x80@\x80p;"\x800\x80\x80\xc8؀\xb\x88\x89\xdx\x93\xf0\xa7\x80
          PKKa\xaciȤ\xc9 
                         /usr/lib/dyld\xaf9\xaflX\xf08ӑ\x98l\xc2\xb4[d2 

4*(\x80@
        8'/usr/lib/libSystem.B.dylib
                                    H/usr/local/opt/libusb/lib/libusb-1.0.0.dylib
                                                                                 h\xff\xa1\x96/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
                                                                      X/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit&X\x92\xf8)P\x93(UH\x89\xe5H\x83\xec H\x89}\xf0H\xc7E\xe8\xb0\xe8\xc5_H\x89E\xe8H\x83\xf8\x84H\x8b\xb8\xe6H\x8b8H\x8d5D˰\xe8G\xc6鑰\xe8\x97H\x89E\xe8H\x83\xf8\x84H\x8b\x86\xe6H\x8b8H\x8d5"˰\xe8\xc6\xe9Z胜H\x89E\xe8H\x83\xf8\x84H\x8bV\xe6H\x8b8H\x8d5
                                                                            ˰\xe8\xe5\xc5\xe9%H\x8b9\xe6H\x8b8H\x8d5
˰\xe8\xc8\xc5H\xc7E\xf8\xe94\xe9\xe9H\x8b}\xe8\xe80H\x83}\xf0\x84H\x8bE\xf0H\x8d
)H\x8H\x8bE\xe8H\x89E\xf8H\x8bE\xf8H\x83\xc4 ]\xc3DUH\x89\xe5H\x83\xec H\x89}\xf0H\x8d\xed(H\x838\x84H\x8d\xdc(H\x83\x85
                   \xc7E\xfc\xfb\xff\xff\xff\xe9\xcbH\x8d\xbe(H\x83x \x85H\x8d\xac(H\x8d
%#H\x89H H\x8d\x9a(H\x83xX\x85H\x8d\x88(H\x8d
\xb1%H\x89HXH\x8dv(H\x83x`\x85H\x8dd(H\x8d
\xfd2H\x89H`H\x8dR(H\x83xx\x85H\x8d@(H\x8d
\x89AH\x89HxH\x8d.(H\x83\xb8\xe8\x85H\x8d(H\x8d
\xd2DH\x89\x88\xe8H\x8d(H\x83\xb8\xf8\x85H\x8d\xef'H\x8d
FH\x89\x88\xf8H\x8d\xda'H\x83\xb8\x88\x85H\x8d\xc5'H\x8d
\xfe4H\x89\x88\x88H\x8d\xb0'H\x83\xb8\x90\x85H\x8d\x9b'H\x8d
48H\x89\x88\x90H\x8d\x86'H\x83\xb8\xa0\x85H\x8dq'H\x8d
\xca6H\x89\x88\xa0H\x8d\'H\x83\xb8\x98\x85H\x8dG'H\x8d
`5H\x89\x88\x98H\x8d2'H\x83\xb8\xa8\x85H\x8d'H\x8d
\x868H\x89\x88\xa8H\x8'H\x83\xb8\x80\x85H\x8d\xf3&H\x8d
lFH\x89\x88\x80H\x8d\xde&H\x83\xb8\xf0\x85H\x8d\xc9&H\x8d
\xb2HH\x89\x88\xf0H\x8d\xb4&H\x83\xb8\x85H\x8d\x9f&H\x8d
JH\x89\x88H\x8d\x8a&H\x83xh\x85H\x8dx&H\x8d
q.H\x89HhH\x8df&H\x83xH\x85H\x8dT&H\x8d
]KH\x89HHH\x8dB&H\x83\xb8\xd0\x85H\x8d-&H\x8d
[... 17 lines cut ...]
\xe9\xe9\x86'
*** [upload] Error 2
==================================== [FAILED] Took 0.24 seconds ====================================

Environment          Status    Duration
-------------------  --------  ------------
ch32v003f4p6_evt_r0  FAILED    00:00:00.246
genericCH32V003A4M6  FAILED    00:00:00.237
=============================== 2 failed, 0 succeeded in 00:00:00.483 ===============================

```

I am unsure if Alex's port uses minichlink as the debug log on Arduino IDE reads like : 
```
Found WCH Link
WCH Programmer is CH32V307 version 2.7
Chip Type: 003
Part Type (A): 0x0010 (This is the capacity code, in KB)
Part UUID    : fe-d9-ab-cd-66-f7-bc-51
PFlags       : ff-ff-ff-ff
Part Type (B): 00-30-05-00
Interface Setup
Image written.

```

IndianTinker, 2023-06-26:
Thanks! It works after replacing the file. I had to do a chmod to remove permissions error.  The only thing is flashes two times instead of just once.

``
Processing ch32v003f4p6_evt_r0 (board: ch32v003f4p6_evt_r0; platform: ch32v; framework: arduino)
-----------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ch32v/ch32v003f4p6_evt_r0.html
PLATFORM: WCH CH32V (1.0.0+sha.131d250) > CH32V003F4P6-EVT-R0
HARDWARE: CH32V003F4P6 48MHz, 2KB RAM, 16KB Flash
DEBUG: Current (wch-link) On-board (wch-link) External (minichlink)
PACKAGES: 
 - framework-arduinoch32v003 @ 0.0.0+sha.cb439a8 
 - tool-minichlink @ 0.1.0+sha.634981a 
 - tool-openocd-riscv-wch @ 2.1100.230329 (11.0) 
 - tool-wchisp @ 0.22.230228 
 - toolchain-riscv @ 1.80200.190731+sha.99cb62f
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/ch32v003f4p6_evt_r0/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.4% (used 376 bytes from 2048 bytes)
Flash: [=         ]   8.7% (used 1420 bytes from 16384 bytes)
Configuring upload protocol...
AVAILABLE: isp, minichlink, wch-link
CURRENT: upload_protocol = minichlink
Uploading .pio/build/ch32v003f4p6_evt_r0/firmware.bin
Found WCH Link
WCH Programmer is CH32V307 version 2.7
Chip Type: 003
Part Type (A): 0x0010 (This is the capacity code, in KB)
Part UUID    : fe-d9-ab-cd-66-f7-bc-51
PFlags       : ff-ff-ff-ff
Part Type (B): 00-30-05-00
Interface Setup
Image written.
``

closed 2023-06-26 by IndianTinker

maxgerhardt, 2023-06-26:
1. Make sure you use the [project environment switcher](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks) to select only **one** environment and not the "Default" one, then only one environment should get triggered
2. That minichlink version can't use line-by-line debugging because it's too old. Either build the latest minichlink version from https://github.com/cnlohr/ch32v003fun/tree/master/minichlink and replace the file again or upgrade the firmware version on the WCHLink, then OpenOCD should work without problems.

francismariano, 2024-01-18:
Hello everyone. I can not also upload the firmware. Here is my log. Could you help me, please?

```
Processing genericCH32V003F4U6 (board: genericCH32V003F4U6; platform: ch32v; framework: noneos-sdk)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ch32v/genericCH32V003F4U6.html
PLATFORM: WCH CH32V (1.1.0+sha.f0d5f28) > Generic CH32V003F4U6
HARDWARE: CH32V003F4U6 48MHz, 2KB RAM, 16KB Flash
DEBUG: Current (wch-link) On-board (wch-link) External (minichlink)
PACKAGES: 
 - framework-wch-noneos-sdk @ 2.20000.0+sha.cdf2322 
 - tool-minichlink @ 0.1.0+sha.af02ba5 
 - tool-openocd-riscv-wch @ 2.1100.230329 (11.0) 
 - tool-wchisp @ 0.22.230228 
 - toolchain-riscv @ 1.80200.190731+sha.99cb62f
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/genericCH32V003F4U6/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.1% (used 288 bytes from 2048 bytes)
Flash: [=         ]   9.0% (used 1480 bytes from 16384 bytes)
Configuring upload protocol...
AVAILABLE: isp, minichlink, wch-link
CURRENT: upload_protocol = wch-link
Uploading .pio/build/genericCH32V003F4U6/firmware.elf
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-03-02-19:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

Warn : Transport "sdi" was already selected
Ready for Remote Connections
Error: open failed

*** [upload] Error 1
========================================================================================== [FAILED] Took 0.41 seconds ==========================================================================================

Environment          Status    Duration
-------------------  --------  ------------
genericCH32V003F4U6  FAILED    00:00:00.414
==================================================================================== 1 failed, 0 succeeded in 00:00:00.414 ====================================================================================

 *  The terminal process "platformio 'run', '--target', 'upload', '--environment', 'genericCH32V003F4U6', '--upload-port', '/dev/cu.usbmodemF3838F0678561'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.
```

Thank you in advanced.

maxgerhardt, 2024-01-18:
>Error: open failed

Sounds like either USB permissions / udev rules are not right or there isn't actually a WCH-Link(E) connected. Can you show the output of `libusb` and how you connected the WCH-LinkE to the CH32V003 chip / board?

maxgerhardt, 2024-01-18:
Can you follow the instructions at https://github.com/Community-PIO-CH32V/platform-ch32v#installation, reboot the computer and try again?

maxgerhardt, 2024-01-18:
My bad. You're on Mac. These Linux instructions make no sense. Let me cross-check the VID and PID on Linux though.

maxgerhardt, 2024-01-18:
Weird. under Linux I see
```
 [  134.914495] usb 1-2: new full-speed USB device number 4 using xhci_hcd
[  135.181434] usb 1-2: New USB device found, idVendor=1a86, idProduct=8010, bcdDevice= 2.09
[  135.181442] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  135.181446] usb 1-2: Product: WCH-Link
[  135.181448] usb 1-2: Manufacturer: wch.cn
[  135.181451] usb 1-2: SerialNumber: 81188F0605F5
```
Which does not match your `Bus 008 Device 005: ID 1a86:8012 1a86 WCH-Link  Serial: F3838F067856` with respect to the product ID.. My adapter explicitly says "WCH-LinkE-RO-1v3" on it.

Another thing you can try is to download the `wlink` tool from https://github.com/ch32-rs/wlink/releases/tag/nightly, unpack it and run
```
 ./wlink  status
19:34:31 [INFO] Connected to WCH-Link v2.9(v29) (WCH-LinkE-CH32V305)
Error: Probe is not attached to an MCU, or debug is not enabled. (hint: use wchisp to enable debug)
```

francismariano, 2024-01-18:
```
$ ./wlink status
Error: WCH-Link is connected, but is not in RV mode
```

maxgerhardt, 2024-01-18:
Mh, you can change it via software too (wlink), but it does recommend the official MounRiver IDE for that:
```
$ ./wlink  mode-switch --rv
19:44:47 [WARN] This is an experimental feature, better use the WCH-LinkUtility
```
http://www.mounriver.com/download

francismariano, 2024-01-18:
Damnnn SWDIO <--> PD1 is wrong in project. TOmorrow I will fix and retrying the upload the firmware. Tks a lot again.

francismariano, 2024-01-19:
All right here. Thank you so much for your help.