Community-PIO-CH32V/platform-ch32v #33: tool-minichlink not installing because of wrong branch name

xsrf, 2023-09-02:
Hey,
I just wanted to try a simple blink example from the PlatformIO examples but it won't upload because it cannot install tool-minichlink. Probably because it's branch now is called `main`, not `master`.
```
Executing task: C:\Users\Andreas\.platformio\penv\Scripts\platformio.exe run --target upload --environment genericCH32V003A4M6 

Processing genericCH32V003A4M6 (board: genericCH32V003A4M6; platform: ch32v; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing git+https://github.com/Community-PIO-CH32V/tool-minichlink.git#master
git version 2.30.2.windows.1
Cloning into 'C:\Users\Andreas\.platformio\.cache\tmp\pkg-installing-xpad0dja'...
warning: Could not find remote branch master to clone.
fatal: Remote branch master not found in upstream origin
VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'master', 'https://github.com/Community-PIO-CH32V/tool-minichlink.git', 'C:\\Users\\Andreas\\.platformio\\.cache\\tmp\\pkg-installing-xpad0dja']
```

maxgerhardt, 2023-09-02:
That was already fixed 3 months ago.

https://github.com/Community-PIO-CH32V/platform-ch32v/commit/131d250034d057222b27b8b75cb557290e333845

Please delete any `C:\Users\Andreas\.platformio\platforms\ch32v*` folder and rebuild the project.

xsrf, 2023-09-02:
Well, that worked. Thx. 
PIO didn't show me any updates for the Platform though, tried that before.

closed 2023-09-02 by xsrf