Community-PIO-CH32V/platform-ch32v #34: Error: checksum mismatch (on Windows)

chihiro1234567, 2023-10-01:
I have encountered the following error 
when uploading a sample program to CH32V203K8T6 using PlatformIO.

[Error]
Error: checksum mismatch - attempting binary compare

[Connection]
Windows11 Host => WCH-LinkE => CH32V203K8T6

[Sample]
https://github.com/Community-PIO-CH32V/ch32-pio-projects
blinky-none-os

* Occasional success.
* The problem does not occur when the target microcontroller is CH32V003
* No problem when run on Ubuntu machine 

Do anyone have any solutions ?
thanks

```
Processing genericCH32V203K8T6 (board: genericCH32V203K8T6; platform: ch32v; framework: freertos)
------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ch32v/genericCH32V203K8T6.html
PLATFORM: WCH CH32V (1.0.0+sha.60ac59a) > Generic CH32V203K8T6
HARDWARE: CH32V203K8T6 144MHz, 20KB RAM, 64KB Flash
DEBUG: Current (wch-link) On-board (wch-link) External (minichlink)
PACKAGES:
 - framework-wch-freertos @ 2.10000.0+sha.5a99a7c
 - framework-wch-noneos-sdk @ 2.10000.0+sha.2cafe78
 - tool-minichlink @ 0.1.0+sha.af02ba5
 - tool-openocd-riscv-wch @ 2.1100.230329 (11.0)
 - tool-wchisp @ 0.22.230228
 - toolchain-riscv @ 1.80200.190731+sha.8ee4117
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio\build\genericCH32V203K8T6\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=======   ]  73.9% (used 15144 bytes from 20480 bytes)
Flash: [==        ]  23.0% (used 15100 bytes from 65536 bytes)
Configuring upload protocol...
AVAILABLE: isp, minichlink, wch-link
CURRENT: upload_protocol = wch-link
Uploading .pio\build\genericCH32V203K8T6\firmware.elf
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-01-03-10:00)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

Warn : Transport "sdi" was already selected
Ready for Remote Connections
[wch_riscv.cpu.0] Target successfully examined.
** Programming Started **
** Programming Finished **
** Verify Started **
Error: checksum mismatch - attempting binary compare
embedded:startup.tcl:1162: Error: ** Verify Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 1223
at file "embedded:startup.tcl", line 1162
*** [upload] Error 1
===== [FAILED] Took 7.75 seconds =

Environment          Status    Duration
-------------------  --------  ------------
genericCH32V203K8T6  FAILED    00:00:07.750
```
[image]

maxgerhardt, 2023-10-01:
>https://github.com/Community-PIO-CH32V/ch32-pio-projects
>blinky-none-os

But the build logs says `framework = freertos`? Can you try only `blinky-none-os`?

In the project tasks

[image]

can you execute "Disable Flash protection" and "Erase flash" successfully?

chihiro1234567, 2023-10-01:
The logs and samples did not match, but the results are the same for blinky-none-os.

Both "Disable Flash Protection" and "Erase Flash" succeed. ( It will not fail. )
But "Upload" is failed. (About one out of ten times it succeeds.)

Disable Flash Protection

```
platformio.exe run --target disable_flash_protection --environment genericCH32V203K8T6 

Processing genericCH32V203K8T6 (board: genericCH32V203K8T6; platform: ch32v; framework: noneos-sdk)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ch32v/genericCH32V203K8T6.html
PLATFORM: WCH CH32V (1.0.0+sha.60ac59a) > Generic CH32V203K8T6
HARDWARE: CH32V203K8T6 144MHz, 20KB RAM, 64KB Flash
DEBUG: Current (wch-link) On-board (wch-link) External (minichlink)
PACKAGES: 
 - framework-wch-noneos-sdk @ 2.10000.0+sha.2cafe78 
 - tool-openocd-riscv-wch @ 2.1100.230329 (11.0) 
 - toolchain-riscv @ 1.80200.190731+sha.8ee4117
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Disabling Flash Protection
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-01-03-10:00)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
debug_level: 2

Info : WCH-LinkE  mode:RV version 2.9 
Info : wlink_init ok
Info : clock speed 6000 kHz
Info : [wch_riscv.cpu.0] datacount=2 progbufsize=8
Info : [wch_riscv.cpu.0] Examined RISC-V core; found 1 harts
Info : [wch_riscv.cpu.0]  XLEN=32, misa=0x40901105
[wch_riscv.cpu.0] Target successfully examined.
Info : gdb port disabled
Info : device id = 0x7650abcd
Info : flash size = 64kbytes
flash 'wch_riscv' found at 0x00000000

Info : Success to Disable Read-Protect
cleared protection for sectors 0 through 63 on flash bank 0

shutdown command invoked
========================================================================= [SUCCESS] Took 1.17 seconds =========================================================================

Environment          Status    Duration
-------------------  --------  ------------
genericCH32V203K8T6  SUCCESS   00:00:01.172
```
Erase Flash

```
platformio.exe run --target erase --environment genericCH32V203K8T6 

Processing genericCH32V203K8T6 (board: genericCH32V203K8T6; platform: ch32v; framework: noneos-sdk)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ch32v/genericCH32V203K8T6.html
PLATFORM: WCH CH32V (1.0.0+sha.60ac59a) > Generic CH32V203K8T6
HARDWARE: CH32V203K8T6 144MHz, 20KB RAM, 64KB Flash
DEBUG: Current (wch-link) On-board (wch-link) External (minichlink)
PACKAGES: 
 - framework-wch-noneos-sdk @ 2.10000.0+sha.2cafe78 
 - tool-openocd-riscv-wch @ 2.1100.230329 (11.0) 
 - toolchain-riscv @ 1.80200.190731+sha.8ee4117
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Erasing Flash
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-01-03-10:00)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
debug_level: 2

Info : WCH-LinkE  mode:RV version 2.9 
Info : wlink_init ok
Info : clock speed 6000 kHz
Info : [wch_riscv.cpu.0] datacount=2 progbufsize=8
Info : [wch_riscv.cpu.0] Examined RISC-V core; found 1 harts
Info : [wch_riscv.cpu.0]  XLEN=32, misa=0x40901105
[wch_riscv.cpu.0] Target successfully examined.
Info : gdb port disabled
Info : device id = 0x7650abcd
Info : flash size = 64kbytes
flash 'wch_riscv' found at 0x00000000

erased sectors 0 through 63 on flash bank 0 in 0.044059s

shutdown command invoked
========================================================================= [SUCCESS] Took 0.75 seconds =========================================================================

Environment          Status    Duration
-------------------  --------  ------------
genericCH32V203K8T6  SUCCESS   00:00:00.755
```

maxgerhardt, 2023-10-01:
That seems very unusual. Can you install the original MounRiver IDE from http://www.mounriver.com/download? When flashing a firmware, it should try and search for WCH-Link(E) updates so you can install them.

chihiro1234567, 2023-10-01:
Thanks for the advice.

I understand the problem in my environment.
It was the USB hub that was causing the problem.

Maybe I am picking up some noise.
Can you adjust the flash speed or something?

chihiro1234567, 2023-10-01:
Connecting directly to the host computer no longer fails.

maxgerhardt, 2023-10-01:
The clock speed only affects the signal speed between the WCH-Link and the target chip, so if the problem was between the USB port and the WCH-Link, it should not be able to affect it..

OpenOCD should accept `-c "adapter_speed 500"` to set the speed to e.g. 500kHz, but currentlty there is no way to set this argument through the `platformio.ini`. It would only be accepted when starting a debug session and using  [`debug_speed = 500`](https://docs.platformio.org/en/latest/projectconf/sections/env/options/debug/debug_speed.html).

closed 2023-10-01 by maxgerhardt

chihiro1234567, 2023-10-01:
I understand.