Community-PIO-CH32V/platform-ch32v #35: Unable to build with arduino framework for CH32V003

Muoshy, 2023-10-07:
Hi!

Thanks for the great work on this platform. I have some issues building using the arduino framework.
Installed according to https://pio-ch32v.readthedocs.io/en/latest/installation.html
Cloned the example repo and opened "blinky-arduino-ch32v003" in PIO.

Straight out of the box, I get an include error for "arduino.h".

```
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (C:\Users\qdaniel\Documents\PlatformIO\Projects\ch32-pio-projects\blinky-arduino-ch32v003\src\main.cpp).C/C++(1696)
cannot open source file "pins_arduino.h" (dependency of "Arduino.h")C/C++(1696)
```
Tried building anyway. Got the following error:

```
In file included from C:\Users\qdaniel\.platformio\packages\framework-arduinoch32v003\cores\arduino\hal\wch-hal-i2c.c:1:
C:\Users\qdaniel\.platformio\packages\framework-arduinoch32v003\cores\arduino/Arduino.h:25:10: fatal error: pins_arduino.h: No such file or directory
```

Found "pins.arduino.h" and included it manually through build flags in platform.ini.

```
build_flags =
    -I"${platformio.packages_dir}/framework-arduinoch32v003/variants/CH32V003"
```

Tried building again and "ch32v003fun.h" was missing this time. Added it to the build flags and built again:
```
build_flags =
    -I"${platformio.packages_dir}/framework-arduinoch32v003/variants/CH32V003"
    -I"${platformio.packages_dir}/framework-arduinoch32v003/ch32v003fun/ch32v003fun"
```
No include errors anymore but seems to be a problem with the linking. 
```
Linking .pio\build\ch32v003f4p6_evt_r0\firmware.elf
c:/users/qdaniel/.platformio/packages/toolchain-riscv/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld.exe: cannot open linker script file C:\Users\qdaniel\.platformio\packages\framework-arduinoch32v003\cores\arduino\ch32v003fun\ch32v003fun-stack.ld: Invalid argument
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\ch32v003f4p6_evt_r0\firmware.elf] Error 1
```
Checked "C:\Users\qdaniel\.platformio\packages\framework-arduinoch32v003\cores\arduino\ch32v003fun\ch32v003fun-stack.ld:" and neither the linker script file nor the "ch32v003fun" folder exists for me. Not sure where to continue from here on.

I have tried reinstalling platform-ch32v and updating it along with PIO. Still same problem. Examples using none-os framework build and flash without any issues straight out of the box. 

Win10 Version	10.0.19045 Build 19045
PlatformIO 6.1.11
ch32v@1.0.0
framework-arduinoch32v003@0.0.0
framework-wch-noneos-sdk@2.10000.0
tool-openocd-riscv-wch@2.1100.230329
toolchain-riscv@1.80200.190731

maxgerhardt, 2023-10-07:
Ah, the "update branch" I did on the repo https://github.com/Community-PIO-CH32V/arduino-wch32v003/commit/fdac06b35f28b7c194483d20ea9f334a5ec6fe7d broke it. Thanks for reporting the breakage, will fix asap.

However, there's also https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/examples/blinky-arduino which uses the more official OpenWCH core. This one *should* work, but the generated binaries are a bit on the heavier side.

maxgerhardt, 2023-10-07:
Fixed in https://github.com/Community-PIO-CH32V/platform-ch32v/commit/91a5c8483c2585fa0a7370f8ecadf669dd5867c5 and https://github.com/Community-PIO-CH32V/arduino-wch32v003/commit/cf873692a7fb2da67f74bbcc93450bad2b6d5f0d. Waiting for CI to confirm.

maxgerhardt, 2023-10-07:
CI confirmed good run.

To get to a clean updated state, I would just recommend to 

1. Close VSCode
2. Delete `C:\Users\qdaniel\.platformio\packages\framework-arduinoch32v003`
3. Delete `C:\Users\qdaniel\.platformio\platforms\ch32v`
4. Restart VSCode
5. Build project again

closed 2023-10-07 by maxgerhardt