Community-PIO-CH32V/platform-ch32v #38: Can add frameworks code in PlatformIO ?

Witawat, 2023-11-01:
Can add frameworks code in PlatformIO ?

this code style arduino how to add support on PlatformIO ?

https://github.com/MoonFox2006/Ardu_CH32V003

can share tutorial for add ?

thankyou

maxgerhardt, 2023-11-01:
This core seems very incomplete. The current https://github.com/openwch/arduino_core_ch32/ core we support for CH32V003 ([example](https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/examples/blinky-arduino)) seems way better for this, has at least all the features that the MoonFox2006 has and more.

Witawat, 2023-11-01:
> This core seems very incomplete. The current https://github.com/openwch/arduino_core_ch32/ core we support for CH32V003 [...]

i thing easy use style  like arduino on wch code and support pwm

maxgerhardt, 2023-11-01:
The OpenWCH Arduino core supports all common Arduino functions and PWM too, using [`analogWrite()`](https://github.com/openwch/arduino_core_ch32/blob/main/cores/arduino/wiring_analog.c#L143-L185). Even the PWM frequency is [adjustable](https://github.com/openwch/arduino_core_ch32/blob/f14a216b441e03907d2c0c9afab03e3985ab6852/cores/arduino/wiring_analog.c#L97-L101).

Witawat, 2023-11-01:
> The OpenWCH Arduino core supports all common Arduino functions and PWM too, using [`analogWrite()`](https://github.com/openwch/arduino_core_ch32/blob/main/cores/arduino/wiring_analog.c#L143-L185). [...]

openwch can support now ?
can check ?

maxgerhardt, 2023-11-01:
You are right, PWM does actually not work. The code is by default not activated (`build_flags = -DTIM_MODULE_ENABLED` needed), but even then, it does not complete since it's incomplete. There is no implementation of HardwareTimer.h/.cpp and a lot of type definitions are missing. IT looks to like this is half-copied from the STM32 Arduino core but not actually finished. You are right to open https://github.com/openwch/arduino_core_ch32/issues/31.

Witawat, 2023-11-05:
> You are right, PWM does actually not work. The code is by default not activated ( needed), but even then, it does not complete since it's incomplete. There is no implementation of HardwareTimer.h/.cpp and a lot of type definitions are missing. IT looks to like this is half-copied from the STM32 Arduino core but not actually finished. You are right to open [openwch/arduino_core_ch32#31](https://github.com/openwch/arduino_core_ch32/issues/31).`build_flags = -DTIM_MODULE_ENABLED`

i found new code to small can add ?

https://github.com/wagiminator/MCU-Templates/tree/main/CH32V003

closed 2024-06-29 by maxgerhardt