Community-PIO-CH32V/platform-ch32v #40: Can the C language format be set to the C99 version?

Witawat, 2023-11-02:
I've tried migrating code from MRS, but it runs on C99, causing some code to not compile. correctly

[image]

Now I've tested three IDEs to find the one that's easiest to use.
1. MRS
2. Clion
3. VSCode

But there's a problem with C99.

maxgerhardt, 2023-11-02:
I thought in C you always had to use
```
#include <stdbool.h>
```
to get the `bool` type. Per https://en.cppreference.com/w/c/types/boolean.

Witawat, 2023-11-03:
[image]

this not support print %f ?

closed 2023-11-04 by Witawat