Community-PIO-CH32V/platform-ch32v #46: CH32X035 PD SNK example not compiling

RathBee, 2024-01-30:
Steps to reproduce:
1.Download CH32X035EVT.zip from WCH's website, extract EVT/EXAM/USBPD/USBPD_SNK/User
2.place all source code in pio /src folder
3.pio build

Expected behavior:
Code complies without any issue

Actual behavior:

Compilation fails.

Error log:

`CONFIGURATION: https://docs.platformio.org/page/boards/ch32v/ch32x035f8u6_evt_r0.html
PLATFORM: WCH CH32V (1.1.0+sha.9ed15cc) > CH32X035F8U6-EVT-R0
HARDWARE: CH32X035F8U6 144MHz, 20KB RAM, 62KB Flash
DEBUG: Current (wch-link) On-board (wch-link) External (minichlink)
PACKAGES: 
 - framework-wch-noneos-sdk @ 2.20000.0+sha.cdf2322 
 - tool-openocd-riscv-wch @ 2.1100.230329 (11.0) 
 - toolchain-riscv @ 1.80200.190731+sha.e8e7ba9
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/ch32x035f8u6_evt_r0/src/PD_Process.o
Compiling .pio/build/ch32x035f8u6_evt_r0/src/main.o
In file included from src/PD_Process.c:15:
src/PD_Process.h:23:8: error: unknown type name 'UINT8'
 extern UINT8  Tmr_Ms_Cnt_Last;
        ^~~~~
src/PD_Process.h:24:8: error: unknown type name 'UINT8'
 extern UINT8  Tmr_Ms_Dlt;
        ^~~~~
src/PD_Process.h:25:8: error: unknown type name 'UINT8'
 extern UINT8  Tim_Ms_Cnt;
        ^~~~~
src/PD_Process.h:27:8: error: unknown type name 'UINT8'
 extern UINT8  PDO_Len;
        ^~~~~
src/PD_Process.h:28:8: error: unknown type name 'PD_CONTROL'
 extern PD_CONTROL PD_Ctl;
        ^~~~~~~~~~
src/PD_Process.h:30:8: error: unknown type name 'UINT8'
 extern UINT8 send_data[ ];
        ^~~~~
src/PD_Process.h:31:8: error: unknown type name 'UINT8'
 extern UINT8 PD_Ack_Buf[ ];
        ^~~~~
src/PD_Process.h:33:37: error: unknown type name 'UINT8'
 extern __attribute__ ((aligned(4))) UINT8 PD_Rx_Buf[ 34 ];
                                     ^~~~~
src/PD_Process.h:34:37: error: unknown type name 'UINT8'
 extern __attribute__ ((aligned(4))) UINT8 PD_Tx_Buf[ 34 ];
                                     ^~~~~
src/PD_Process.h:44:8: error: unknown type name 'UINT8'
 extern UINT8 PD_Detect( void );
        ^~~~~
src/PD_Process.h:46:29: error: unknown type name 'UINT8'; did you mean 'UINT8_C'?
 extern void PD_Load_Header( UINT8 ex, UINT8 msg_type );
                             ^~~~~
                             UINT8_C
src/PD_Process.h:46:39: error: unknown type name 'UINT8'; did you mean 'UINT8_C'?
 extern void PD_Load_Header( UINT8 ex, UINT8 msg_type );
                                       ^~~~~
                                       UINT8_C
src/PD_Process.h:47:8: error: unknown type name 'UINT8'
[... 691 lines cut ...]
src/PD_Process.c:786:39: error: 'DEF_TYPE_GET_STATUS_R' undeclared (first use in this function)
                 PD_Load_Header( 0x01, DEF_TYPE_GET_STATUS_R );
                                       ^~~~~~~~~~~~~~~~~~~~~
src/PD_Process.c:790:18: error: 'DEF_TYPE_VCONN_SWAP' undeclared (first use in this function)
             case DEF_TYPE_VCONN_SWAP:
                  ^~~~~~~~~~~~~~~~~~~
src/PD_Process.c:792:39: error: 'DEF_TYPE_REJECT' undeclared (first use in this function)
                 PD_Load_Header( 0x00, DEF_TYPE_REJECT );
                                       ^~~~~~~~~~~~~~~
src/PD_Process.c:796:18: error: 'DEF_TYPE_VENDOR_DEFINED' undeclared (first use in this function)
             case DEF_TYPE_VENDOR_DEFINED:
                  ^~~~~~~~~~~~~~~~~~~~~~~
src/PD_Process.c:809:31: error: request for member 'Flag' in something not a structure or union
                         PD_Ctl.Flag.Bit.VDM_Version = 0;
                               ^
src/PD_Process.c:813:31: error: request for member 'Flag' in something not a structure or union
                         PD_Ctl.Flag.Bit.VDM_Version = 1;
                               ^
src/PD_Process.c:827:15: error: request for member 'Flag' in something not a structure or union
         PD_Ctl.Flag.Bit.Msg_Recvd = 0;                                    /* Clear the received flag */
               ^
src/PD_Process.c:828:15: error: request for member 'PD_BusIdle_Timer' in something not a structure or union
         PD_Ctl.PD_BusIdle_Timer = 0;                                      /* Idle time cleared */
               ^
*** [.pio/build/ch32x035f8u6_evt_r0/src/PD_Process.o] Error 1`

maxgerhardt, 2024-01-30:
You are right, this does not work in this platform currently because we are using `V1.0.0` of the CH32X035 SPL, while the latest one is `V1.0.1` and has the USB-PD header and code. I will update the base package shortly.

maxgerhardt, 2024-01-30:
Fixed in https://github.com/Community-PIO-CH32V/framework-wch-noneos-sdk/commit/1ee8c787bcda01cf5130f878b01022437404c4f2. This added USB-PD support to the SPL. 

I've also added the USBPD-SINK project into our examples, test-compiling them in the CI works fine (https://github.com/Community-PIO-CH32V/platform-ch32v/commit/693fb7a5381e7f0e75dcba9a7a141161b4655031).

closed 2024-01-30 by maxgerhardt