Community-PIO-CH32V/platform-ch32v #48: Zephyr Support

LeoDJ, 2024-02-13:
Hi, I was just asking if there are any plans to support the Zephyr framework sometimes in the future?

There's currently a WIP implementation for the CH32X035 over here: https://github.com/dragonlock2/zephyrboards
(Blog article about what was done: https://matthewtran.dev/2024/02/basic-ch32v-zephyr-support/ )

I've looked at it for a few hours, but it appears that I'm sadly not well versed enough in the PlatformIO package / framework ecosystem to include it myself.
I'm unsure how to merge/overlay the custom drivers with the official `framework-zephyr` and then include that into this CH32 platform. Or how the framework-zephyr package works at all. I couldn't even find a GitHub repo for it, PIO just downloads it from its own opaque repository as far as I can tell?
It's all a bit convoluted to me of where to best get started...

maxgerhardt, 2024-02-13:
Thanks for the link to the blog post, I will look into it. I've worked once with the Zephyr package. Indeed, `framework-zephyr` is not in a PlatformIO repo, they just add the magic-sauce (the `scripts/platformio/platformio-build.py`) themselves to Zephyr when the package is upload. Since many PlatformIO platforms have Zephyr support (ststm32, nordicnrf52, ...), and also these "Zephyr modules" are managed (which is what the `zephyrboards` repo looks to be), this should be doable.

maxgerhardt, 2024-02-14:
While pulling in the CH32 boards as a Zephyr module failed to work for me (it added the module to the build but failed to recognize the new boards it provided), I've just added the repo on top of Zephyr 3.5.0 with some slight modifications and additions: https://github.com/Community-PIO-CH32V/framework-zephyr. Now it does find the `usb_pdmon` board, but the CH32 HAL module is not added yet. If that is solved, it should be very close to working and the `zephyr` branch here can be merged.

maxgerhardt, 2024-02-14:
Compilation of the first empty Zephyr program for CH32X035 has been achieved.
```
Memory region         Used Size  Region Size  %age Used
             ROM:       20948 B        62 KB     33.00%
             RAM:        7332 B        20 KB     35.80%
        IDT_LIST:          0 GB         2 KB      0.00%
MethodWrapper(["checkprogsize"], [".pio\build\usb_pdmon\firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.0% (used 0 bytes from 20480 bytes)
Flash: [          ]   0.0% (used 0 bytes from 63488 bytes)
```

maxgerhardt, 2024-02-17:
Added Zephyr support for the `usb_pdmon_ch32x035g8u6` board per PR above. Tested on a generic CH32X035 dev board. Works fine with blinky + serial output.

```
--- Terminal on COM13 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Toggled LED! 4
Toggled LED! 5
Toggled LED! 6
..
```

PR #49 (2024-02-17): Add Zephyr support
Fixes #48
closed 2024-02-17 by maxgerhardt