Community-PIO-CH32V/platform-ch32v #51: [upload] Error 3221225477

LovelyA72, 2024-02-24:
I am not able to upload a fresh project to my CH32V307VCT6 board

my config:
```
[env:uno]
platform = https://github.com/Community-PIO-CH32V/platform-ch32v.git
board = genericCH32V307VCT6 
framework = arduino
```

upload log (partial):
```
Configuring upload protocol...
AVAILABLE: isp, minichlink, wch-link
CURRENT: upload_protocol = wch-link
Uploading .pio\build\uno\firmware.elf
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-09-22-10:36)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

Warn : Transport "sdi" was already selected
Ready for Remote Connections
*** [upload] Error 3221225477
```

maxgerhardt, 2024-02-24:
I have seen OpenOCD crash when an WCH-Link was used that was not in RISC-V mode, or had the latest firmware installed on it via the MounRiver IDE.

Can you show a screenshot of the device manager with "View -> Device By Container" of your WCH-Link? In the "details" view, what PID and VID does it have? 

E.g.,
[image]
[image]

LovelyA72, 2024-02-24:
After installing the driver it stopped hard crashing. But it still can not upload.
```
Uploading .pio\build\uno\firmware.elf
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-09-22-10:36)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

Warn : Transport "sdi" was already selected
Ready for Remote Connections
[wch_riscv.cpu.0] Target successfully examined.
** Programming Started **
Error: error writing to flash at address 0x00000000 at offset 0x00000000
embedded:startup.tcl:1162: Error: ** Programming Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 1227
at file "embedded:startup.tcl", line 1162
*** [upload] Error 1
```

maxgerhardt, 2024-02-24:
Seems like https://github.com/Community-PIO-CH32V/platform-ch32v?tab=readme-ov-file#important-notices to me then.

LovelyA72, 2024-02-24:
Thank you so much! It now works properly. It will still be nice to be able to exit gracefully with error message when there's a driver issue instead of hard crashing.

closed 2024-02-24 by LovelyA72

maxgerhardt, 2024-02-24:
Graceful exit would indeed be nice, however that's something WCH / MounRiver must fix, since those are their binaries. They also don't upstream their OpenOCD changes yet (although it violates the GPL), and source code releases appear only very sparingly.

maxgerhardt, 2024-02-26:
This already works using `upload_protocol = minichlink`, I tested this with a WCH-Link (loaded WinUSB drivers via Zadig on Windows) and a CH32V003. That tool might however not support all chips, and its GDB is still not properly usable from within PlatformIO (#28).

```
CURRENT: upload_protocol = minichlink
minichlink -w .pio\build\ch32v003f4p6_evt_r0\firmware.bin 0x08000000 -b
Interface Setup
Image written.
Found WCH Link
WCH Programmer is LinkE version 2.11
Chip Type: 003
Setup success
Flash Storage: 16 kB
Part UUID    : ab-c8-ab-cd-13-de-bc-49
PFlags       : ff-ff-ff-ff
Part Type (B): 07-13-bb-91
Read protection: disabled
====== [SUCCESS] Took 6.73 seconds ======
```