Community-PIO-CH32V/platform-ch32v #53: ch582 3 mode keyboard example build error

kahoch, 2024-03-10:
I got a 3 mode keyboard example from WCH, it can be built by MRS, but I got a lot of "undefined reference to xxx" error in PIO. Just show as below.

> Processing genericCH582M (platform: ch32v; board: genericCH582M; framework: noneos-sdk)
> -----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ch32v/genericCH582M.html
> PLATFORM: WCH CH32V (1.0.0+sha.f515d0a) > Generic CH582M
> HARDWARE: CH582M 20MHz, 32KB RAM, 480KB Flash
> DEBUG: Current (wch-link) On-board (wch-link) External (minichlink)
> PACKAGES:
>  - framework-wch-noneos-sdk @ 2.10000.0+sha.5992d69
>  - tool-openocd-riscv-wch @ 2.1100.230922+sha.718adf8
>  - tool-wchisp @ 0.22.230228
>  - toolchain-riscv @ 1.80200.190731+sha.8ee4117
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 12 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- HAL
> |-- BLE_LIB
> Building in release mode
> Compiling .pio\build\genericCH582M\src\BLE\hiddev.o
> ...
> ...
> Linking .pio\build\genericCH582M\firmware.elf
> c:/users/kaho/.platformio/packages/toolchain-riscv/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld.exe: .pio\build\genericCH582M\src\main.o: in function `Main_Circulation':
> D:\MyProjects\CH582\kb2/src/main.c:40: undefined reference to `rf_tx_process'
> c:/users/kaho/.platformio/packages/toolchain-riscv/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld.exe: D:\MyProjects\CH582\kb2/src/main.c:41: undefined reference to `TMOS_SystemProcess'
> c:/users/kaho/.platformio/packages/toolchain-riscv/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld.exe: .pio\build\genericCH582M\src\main.o: in function `main':
> ...
> a lot of similar error
> ...
> collect2.exe: error: ld returned 1 exit status
> *** [.pio\build\genericCH582M\firmware.elf] Error 1

I have re-organized the folder structure, put .h files in the "include" folder, .c files in the "src" folder and the library files in the "lib" folder.

here is my platform.ini:

> [env:genericCH582M]
> platform = ch32v
> board = genericCH582M
> framework = noneos-sdk
> ; upload_protocol = isp
> upload_protocol = wch-link
> monitor_speed = 115200
> build_flags = 
>     -DCONFIG_RF_MODE=1 
>     -DCLK_OSC32K=1
>     -DCONFIG_RF_HOP
>     -DCONFIG_RF_PAIR
>     -DCONFIG_RF_PAIR_DEBUG
>     -DCONFIG_RF_PRC_DEBUG
>     -DCONFIG_RF_RESEND
>     -DCONFIG_RISCV
>     -DLOG
>     -DDEBUG=1
>     -I./test
>     -I./include
>     -I./include/BLE
>     -I./include/RF_task
>     -I./include/sys
>     -I./include/toolchain
>     -I./include/USB_task
>     -I./lib
>     -I./lib/HAL/include
>     -I./lib/backlight
>     -I./lib/BLE_LIB
>     -I./lib/debug
>     -I./lib/dfu
>     -I./lib/HAL_FLASH
>     -I./lib/I2C
>     -I./lib/key_scan
>     -I./lib/RF_PHY
>     -I./lib/RingBuffer
>     -I./lib/USB
>     -I./lib/soc

How can I solve this problem? Thanks!

maxgerhardt, 2024-03-10:
If `TMOS_SystemProcess` is undefined, where is this function implemented? If it's in a source file (`*.c`, `*.S`, ..), then that file **must** be either in `src/` or in `lib/<subfolder>`. If you cannot find the function in any such file, it may be defined by some `libXXX.a` library which you forgot to link in (`-L<path to folder where the .a is>, `-l<name of library>`).

If you post the **exact** project files, we can help you more.

kahoch, 2024-03-10:
Acturally all of these functions are defined, in vscode, I can use "go to difinition" to trace them. 
Please check the whole project I uploaded. Please note there is a modified CH58x_common.h file.
[kb2.zip](https://github.com/Community-PIO-CH32V/platform-ch32v/files/14550490/kb2.zip)

kahoch, 2024-03-10:
And this is the original WCH example.
[Three-mode_kbd.zip](https://github.com/Community-PIO-CH32V/platform-ch32v/files/14550519/Three-mode_kbd.zip)

maxgerhardt, 2024-03-10:
[image]

No, they are **declared** (to exist), not **defined** in any C file. The file that defines them is in the same folder as the header file.

[image]

What you are missing is to include those `.a` files (linISP583 should be autoincluded by the SPL already).
1. Rename `LIBCH58xBLE.a` to just `libCH58xBLE.a` (will prevent build errors on non-Windows systems)
2. Add 
```
    -Llib/BLE_LIB/
    -lCH47xBLE
```
to the `build_flags`
3. Rebuild.

A similiar thing is done is the official example here: <https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/examples/ble-usb-cdc-ch58x/lib/BLE_LIB>. There, the inclusion of the `.a` file is done through the `library.json` automatically.

maxgerhardt, 2024-03-10:
Due to this project having a modified `CH58x_common.h` (bad practice! this is supposed to be standard and the same for all SDK projects!), it's easier to **not** use `framework = nonesos-sdk` and go with bare-metal instead. The project contains all needed files anyways. It's also easier to **not** use "PlatformIO-style libraries" in `lib/` because the library dependency finder fails on the files with the weird chinese encoding and can't detect proper dependency chains, even with `lib_ldf_mode = chain+`.

This here is the converted version of the project. Not a single line of C source code has been touched. Only the configuration in terms of defined macros, include paths and linked libraries were copied between the MounRiverStudio project and PlatformIO.

[Three-mode_kbd_conv.zip](https://github.com/Community-PIO-CH32V/platform-ch32v/files/14550672/Three-mode_kbd_conv.zip)

kahoch, 2024-03-10:
Thank you!!! I'm really appreciate your help and your incredible work! I have been struggling with this problem for few days.

closed 2024-03-10 by kahoch

maxgerhardt, 2024-03-10:
This problem is mostly lack of documentation on our side. <https://readthedocs.org/projects/pio-ch32v/> is wholly incomplete as of now. I will write a "How to convert a MounRiverStudio project to PlatformIO" page, this would have taken care of all problems here, and will take care of problems in the future.