Community-PIO-CH32V/platform-ch32v #56: CH32V003 Debug Problems Windows

alexandrkolesnikov1, 2024-05-02:
Hello everyone! I'm trying to set up debug on CH32V003. But it turns out that I only see assembler lines and do not see my code during debugging. How can I achieve that I can see the disassembled code?

[image]

 platformio.ini 
``` [env:genericCH32V003F4U6] 
platform = ch32v 
board = genericCH32V003F4U6 
framework = arduino 
build_type = debug
board_build.core = openwch 
monitor_speed = 115200 ```

alexandrkolesnikov1, 2024-05-02:
Note: I use WCH-Link(E)

maxgerhardt, 2024-05-05:
That shouldn't happen. Only initially on function entry, it does go into the `__riscv_save_x` function, but once you carefully step-into each instruction, it should return to the C/C++ code normally. 

Does this also happen with the most simple blinky example? https://github.com/Community-PIO-CH32V/platform-ch32v/blob/develop/examples/blinky-arduino/src/main.cpp

alexandrkolesnikov1, 2024-05-06:
@maxgerhardt Yes it happens with blink example too. But I fix this problem when rewrite my code to noneos-sdk. I dont now why but it happens when i use Arduino sdk and it happens only with users code, for example "ch32v00x_usart.h"/"ch32v00x_usart.c" debug works normaly