Community-PIO-CH32V/platform-ch32v #62: SDI printf

ldab, 2024-06-28:
I'm not too familiar with how things are set up here, but I have a `CH32V003` printing over SDI (one wire) but for that to happen first the "enable SDI Printf" needs to be sent on the WCH-LinkUtility, I can do so on Windows, but on linux the tool is not available:

[image]

It seems tho that a CLI is available here: https://github.com/ch32-rs/wlink#feature-support which seems to support the feature, so theoretically we sent `wlink flash --enable-sdi-print` do you see how this could be integrated on the flasher?

Basically the issues is more like a dumb question to "where/which flasher is used?"

ldab, 2024-06-28:
Ok, now I see, https://github.com/Community-PIO-CH32V/wchisp is a fork of the repo mentioned above, do you think we can sync it?

ldab, 2024-06-28:
So, theoretically I add a new uploader her? https://github.com/Community-PIO-CH32V/platform-ch32v/blob/6dce5eb1338cc96587d70c0bc814b8357378bd5a/platform.json#L142

maxgerhardt, 2024-06-28:
Yes. But that also has to be matched by providing the download URL for PlatformIO compatible packages for that tool (tar.gz packages that have `package.json` + the [platform-specific wlink executable](https://github.com/ch32-rs/wlink/releases/tag/nightly)). `wlink` is not yet added, but can easily be added.

To answer the question: 
>"where/which flasher is used?"

Depending on which `upload_protocol = ...` is selected from the available ones

https://github.com/Community-PIO-CH32V/platform-ch32v/blob/6dce5eb1338cc96587d70c0bc814b8357378bd5a/boards/ch32v307_evt.json#L44-L48

we either use
1. The WCH fork of *OpenOCD*.
2. CNLohr's `minichlink` executable ([here](https://github.com/Community-PIO-CH32V/ch32v003fun/tree/master/minichlink))
3. Or wchisp (only thing that can do USB ISP bootloader uploads now)

Let me just add `wlink` tool support and a platform target for "View SDI Printfs via wlink".,

maxgerhardt, 2024-06-28:
Ahh, I have to correct myself. We don't even need to enable uploading with that tool, just invoking it `wlink` with that `--enable-sdi-print` is going to make the serial port of the WCH-Link spit out the decoded SDI prints automatically, as normal?

maxgerhardt, 2024-06-28:
I've made addition https://github.com/Community-PIO-CH32V/platform-ch32v/commit/a73ed6d664f4cefff840a7f79228122fd82c874f.
* `upload_protocol = wlink` now available for all boards, will use `wlink flash` to flash
* `tool-wlink` now always installed from https://github.com/Community-PIO-CH32V/tool-wlink no matter the uploader
  * so that "Enable SDI Printf" task is available even if `upload_protocol = wch-link` (OpenOCD) is selected, otherwise the tool-wlink package may not be guaranteed to be installed
* task for "Enable SDI Printf" available, triggers `wlink sdi-print enable`

[image]

You'll need to do a full platform update and window reload of VSCode. Easiest is to delete `C:\Users\<user>\.platformio\platforms\ch32v*` and restart VSCode.

ldab, 2024-06-28:
that is super cool, it works no problem:

```
Enabling SDI Printf
14:01:18 [INFO] Connected to WCH-Link v2.10(v30) (WCH-LinkE-CH32V305)
14:01:19 [INFO] Attached chip: CH32V003 [CH32V003F4P6] (ChipID: 0x00300500)
14:01:19 [INFO] Enabling SDI print
14:01:19 [INFO] Now you can connect to the WCH-Link serial port
```

[image]

I think the next step would be to add an option to the monitor itself, something like `monitor_sdi = true` to the ini file, what do you think?

maxgerhardt, 2024-06-28:
Would be nice, but I'm not sure at the moment if the PlatformIO core allows a "pre" hook for the `monitor` task or a prediction thereof, where we could execute a quick `wlink sdi-print enable` if that project option is set..

closed 2024-06-28 by ldab