Community-PIO-CH32V/platform-ch32v #64: Is there any way to make PIO-CH32V work with Winusb?

Rockwell1799, 2024-07-12:
I'm on windows 11 and the vscode with PIO-CH32 is working fine with the CH32V003FUN project,but I want to use the minichlink ,but unfortunately it's only working with Winusb not the proprietary CH-link driver and Vscode with PIO32 doesn't work with Winusb.
So,is there any way to still use Vscode and minichlink at the same time?
TIA

maxgerhardt, 2024-07-12:
Please try the upload protocols
```
upload_protocol = minichlink
```
and
```
upload_protocol = wlink
```

Rockwell1799, 2024-07-12:
Thank you! I guess you meant either minichlink or wlink?

maxgerhardt, 2024-07-12:
Yes. They will use <https://github.com/cnlohr/ch32v003fun/tree/master/minichlink> and <https://github.com/ch32-rs/wlink/> respectively. However, with `minichlink` I have not yet integrated the debugging feature, there are some problems with it; so they can only *upload* properly, not debug.

Rockwell1799, 2024-07-12:
Thank you for your prompt reply!
Let's say if I want to use wlink, in which path do I have to copy wlink.exe?

Rockwell1799, 2024-07-12:
Thank you Max, you are very helpful! 
I'm going to try that when I get back home.

closed 2024-07-15 by Rockwell1799