Community-PIO-CH32V/platform-ch32v #65: ch32v003fun-Error: This board doesn't support ch32v003fun framework!

Rockwell1799, 2024-07-15:
Hello
I'm having a problem with the ch32v003fun example in this repository.My platformIO.ini is the following:
`[env]
platform = ch32v
framework = ch32v003fun

monitor_speed = 115200
; uncomment this to use USB bootloader upload via WCHISP
;upload_protocol = isp
upload_protocol = minichlink

[env:ch32v003f4p6_evt_r0]
board = ch32v003f4p6_evt_r0`

I'm getting this error. What I'm doing wrong?
TIA

`Resolving ch32v003f4p6_evt_r0 dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing ch32v003f4p6_evt_r0 (board: ch32v003f4p6_evt_r0; platform: ch32v; framework: ch32v003fun)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ch32v/ch32v003f4p6_evt_r0.html
PLATFORM: WCH CH32V (1.1.0+sha.55aad82) > CH32V003F4P6-EVT-R0
HARDWARE: CH32V003F4P6 48MHz, 2KB RAM, 16KB Flash
DEBUG: Current (wch-link) On-board (wch-link) External (minichlink)
PACKAGES: 
 - tool-minichlink @ 0.1.0+sha.3d2abd2 
 - tool-openocd-riscv-wch @ 2.1100.230922+sha.718adf8 
 - tool-wlink @ 0.22.240627+sha.030572c 
 - toolchain-riscv @ 1.80200.190731+sha.8ee4117
Error: This board doesn't support ch32v003fun framework!`

maxgerhardt, 2024-07-15:
You're using an outdated version of the platform, the latest version supports it. Please follow the README: https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop?tab=readme-ov-file#updating

https://github.com/Community-PIO-CH32V/platform-ch32v/blob/b7397c29a71101175bfc94f6ab06f9daac336458/boards/ch32v003f4p6_evt_r0.json#L31-L35

Rockwell1799, 2024-07-15:
Thank you, Max. Every thing is working fine, after updating the platorm!

closed 2024-07-28 by maxgerhardt