Community-PIO-CH32V/platform-ch32v #67: CH592F: isp flash failed

KenjutsuGH, 2024-08-15:
When I try to upload to my WeAct CH592F board using the **isp** protocol, I get the error:

```
Configuring upload protocol...
AVAILABLE: isp, minichlink, wch-link, wlink
CURRENT: upload_protocol = isp
Uploading .pio/build/genericCH592F/firmware.elf
Error: Device type of 0x22 not found
*** [upload] Error 1
```

When I go to the project directory and manually invoke **wchisp (0.2.3)**, it works:

```
wchisp  flash .pio/build/genericCH592F/firmware.elf                    
08:53:06 [INFO] Chip: CH592[0x9222] (Code Flash: 448KiB, Data EEPROM: 32KiB)
08:53:06 [INFO] Chip UID: 42-38-4F-10-53-5C-E4-A4
08:53:06 [INFO] BTVER(bootloader ver): 02.30
08:53:06 [INFO] Current config registers: ffffffffffffffff4d0fff4f0002030042384f10535ce4a4
RESERVED: 0xFFFFFFFF
WPROTECT: 0xFFFFFFFF
  [0:0]   NO_KEY_SERIAL_DOWNLOAD 0x1 (0b1)
    `- Enable
  [1:1]   DOWNLOAD_CFG 0x1 (0b1)
    `- PB22(Default set)
USER_CFG: 0x4FFF0F4D
  [2:0]   RESERVED 0x5 (0b101)
    `- Default
  [3:3]   CFG_RESET_EN 0x1 (0b1)
    `- Enable
  [4:4]   CFG_DEBUG_EN 0x0 (0b0)
    `- Disable
  [5:5]   RESERVED 0x0 (0b0)
    `- Default
  [6:6]   CFG_BOOT_EN 0x1 (0b1)
    `- Enable
  [7:7]   CFG_ROM_READ 0x0 (0b0)
    `- Disable the programmer to read out, and keep the program secret
  [27:8]  RESERVED 0xFFF0F (0b11111111111100001111)
    `- Error
  [31:28] VALID_SIG 0x4 (0b100)
    `- Valid
08:53:06 [INFO] Read .pio/build/genericCH592F/firmware.elf as ELF format
08:53:06 [INFO] Found loadable segment, physical address: 0x00000000, virtual address: 0x00000000, flags: 0x5
08:53:06 [INFO] Section names: [".init"]
08:53:06 [INFO] Found loadable segment, physical address: 0x00000004, virtual address: 0x20000000, flags: 0x5
08:53:06 [INFO] Section names: [".highcode"]
08:53:06 [INFO] Found loadable segment, physical address: 0x00000874, virtual address: 0x00000874, flags: 0x5
08:53:06 [INFO] Section names: [".text"]
08:53:06 [INFO] Found loadable segment, physical address: 0x00000ea8, virtual address: 0x20000870, flags: 0x6
08:53:06 [INFO] Section names: [".data"]
08:53:06 [INFO] Firmware size: 4096
08:53:06 [INFO] Erasing...
08:53:06 [WARN] erase_code: set min number of erased sectors to 8
08:53:06 [INFO] Erased 8 code flash sectors
08:53:07 [INFO] Erase done
08:53:07 [INFO] Writing to code flash...
██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 4096/409608:53:07 [INFO] Code flash 4096 bytes written
08:53:08 [INFO] Verifying...
██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 4096/409608:53:08 [INFO] Verify OK
08:53:08 [INFO] Now reset device and skip any communication errors
08:53:08 [INFO] Device reset
```

My installed packages:
 - framework-wch-noneos-sdk @ 2.20000.0+sha.3d1f411 
 - tool-minichlink @ 0.1.0+sha.af02ba5 
 - tool-openocd-riscv-wch @ 2.1100.230922+sha.6eeab87 
 - tool-wchisp @ 0.22.230228 
 - tool-wlink @ 0.22.240627+sha.22bfa43 
 - toolchain-riscv @ 1.80200.190731+sha.99cb62f

My platformio.ini:
```
[env:genericCH592F]
platform = ch32v
board = genericCH592F
framework = noneos-sdk
upload_protocol = isp
```

I am using **macOS Sonoma 14.5 (23F79) M1**

maxgerhardt, 2024-08-16:
>When I go to the project directory and manually invoke wchisp (0.2.3), it works:

Can you copy your latest wchisp version into `~/.platformio/packages/tool-wchisp/`  to overwrite the old version? Does PlatformIO upload then?

KenjutsuGH, 2024-08-17:
Thank you @maxgerhardt, that solves the problem

closed 2024-08-17 by KenjutsuGH

maxgerhardt, 2024-08-18:
Thanks for letting me to, I'll try to update the packages in a scripted way some time in the future.

s00500, 2024-09-07:
Just ran into the same, installed the 2.3 from cargo and copied it, now it works

maxgerhardt, 2024-09-14:
The updated `tool-wchisp` pacakges have been `pio pkg publish`'ed today, but are in manual review by the PlatformIO team before they're visible.