Community-PIO-CH32V/platform-ch32v #68: Ch32V003: Please allow setting option byte register from within the IDE

FPGA-Computer, 2024-09-04:
CH32V003 Option Bytes Register (FLASH_OBR) uses the default setting whenever code is downloaded from the compiler/Debugger environment.  There doesn't seem to be a way of changing the default values from within the IDE.   External WCH utility could be used, BUT the debugger will rewrite it with the default values.

e.g. PD7 is default as NRST pin.  This pin could also be used as a GPIO.  However, there are no IDE options to tell the Debugger to use a different Option Byte value whenever it downloads a new firmware.  The only alternative is the firmware programmatically reprogram the option byte register as part of the user firmware.   However, I have not tested if the changes would affects the GPIO directly without a reset. 

[image]

maxgerhardt, 2024-09-04:
It shouldn't happen at all that a regular upload with OpenOCD or `wlink` or whatever touches the option bytes of the board. With what `platformio.ini` is it? How do you verify that the option bytes have changed?

FPGA-Computer, 2024-09-10:
My picture is a composite of screen shots - left is from within debugger, top is user manual and bottom is drop down menu from WCH utility showing the various options available.

>How do you verify that the option bytes have changed?
Left side of my picture shows the FLASH_OBR register viewed within the Debugger.  It shows the bit fields for RST_MODE that is different than what I programmed in from WCH-Link Utility.

As I debug my code, the IDE downloads a new copy of the firmware.  Since there are no settings available within the IDE, it picks whatever the default state is or whatever the erased state.

>With what platformio.ini
Is there a setting in there I can change that the Option byte that tells the debugger to program it differently?   This is what I am asking for.  If there is a way of doing so, then please share the steps.

LokiMetaSmith, 2025-03-03:
I'm trying to figure this out as well. 

the best I have come up with is to use a special firmware load that you would run once to internally flash the user option register.

One of the problems with doing a generic platform.io update is the option register is used to add security features and is very specific for the ch32, so a pre-load "test" firmware might be necessary to properly configure this. 

WCH-Link Utility is different than the wlink cli tool that platform.io uses. 

There's a good chance that wlink needs to be modified to give it a cli option to set the option register correctly.