Community-PIO-CH32V/platform-ch32v #69: ch592 examples question

s00500, 2024-09-07:
Hey, nice Project!!!
I was able to build and upload some basic blink sketches to an ch592F

Now I know this might not be 100% related to this project... 
But I am trying to get this example moved over to pio... 

https://github.com/openwch/ch592/tree/main/Application/PDF_DataLogger

but I cant make the libpdf.a link properly... I keep running into linker errors.. any tips ?

I tried adding buildflags pointing to the .a file without much luck...

```
Indexing .pio/build/main/libFrameworkNoneOSVariant.a
Linking .pio/build/main/firmware.elf
/Users/lb/.platformio/packages/toolchain-riscv/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld: .pio/build/main/src/pdfTempe.o: in function `pdf_create':
/Users/lb/Documents/PlatformIO/Projects/CH5Test/src/pdfTempe.c:163: undefined reference to `LPDF_InitParam'
/Users/lb/.platformio/packages/toolchain-riscv/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld: /Users/lb/Documents/PlatformIO/Projects/CH5Test/src/pdfTempe.c:186: undefined reference to `LPDF_New'
/Users/lb/.platformio/packages/toolchain-riscv/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld: /Users/lb/Documents/PlatformIO/Projects/CH5Test/src/pdfTempe.c:187: undefined reference to `pdf_template_creat'
/Users/lb/.platformio/packages/toolchain-riscv/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld: /Users/lb/Documents/PlatformIO/Projects/CH5Test/src/pdfTempe.c:187: undefined reference to `LPDF_SaveToFile
```

maxgerhardt, 2024-09-14:
Sorry for the late response. Can you post your current project with the `platformio.ini`? This hsould be as simple as
```ini
build_flags =
   -L<path where libxxxx.a is>
   -l<xxxxx>
```

s00500, 2024-09-17:
[image]
This is what it looks like right now

maxgerhardt, 2024-09-18:
That looks extraordinarily weird. Why is the whole Startup and StdPeriphDriver and RVMSIS lib `lib/`? When you set `framework = noneos-sdk`, the startup file, peripheral code, debug code and system code is already compiled for you, as sourced from https://github.com/Community-PIO-CH32V/framework-wch-noneos-sdk. 

Let me just convert the original project properly to show you the clean way.

maxgerhardt, 2024-09-18:
Example converted and added into https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/examples/usb-pdf-logger-none-os-ch592. 

Pay close attention to the `platformio.ini` and the two created `library.json` files. Pay attention to what files were added and which weren't. Cross-reference everything against the documentation, such as [library.json](https://docs.platformio.org/en/latest/manifests/library-json/index.html) and the [LDF modes](https://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode).

s00500, 2024-09-19:
@maxgerhardt Exceptional! 

Why was it done like I did it ? cause I did not know what I was doing ;-)

Thanks for including the example!

s00500, 2024-09-19:
Falls es bei euch einen Link für die Kaffeekasse gibt wäre ich interessiert ;-)