Community-PIO-CH32V/platform-ch32v #75: Fatal error when building blinky on CH32X033

Rockwell1799, 2024-11-14:
Hello
I'm using the examples on this repository below, but I keep getting a fatal error regarding a debug.h file missing. I have used other examples on CH32V003 and they are all working fine.
Can someone help me please?

https://github.com/wagiminator/Development-Boards/tree/main/CH32X033F8P6_DevBoard/software/blink

[image]

Rockwell1799, 2024-11-14:
Never mind! By removing the line debug.h, everything is working fine!

maxgerhardt, 2024-11-14:
This is still supposed to work because the CH32X035 SDK (also used for X033) **does** have the `debug.h`.

https://github.com/Community-PIO-CH32V/framework-wch-noneos-sdk/tree/main/Debug/ch32x035

Let me check it out..

maxgerhardt, 2024-11-14:
Wtf?

Why is this set to `no`, so that the debug directory and code won't be added, but still `debug.h` is included? 

https://github.com/wagiminator/Development-Boards/blob/0c34956e895d184507ce73b2b2f01aaf6da6cf62/CH32X033F8P6_DevBoard/software/blink/platformio.ini#L24

maxgerhardt, 2024-11-14:
Mhmm I see. The include doesn't come from the *firmware code*, but from the auto-included SDK code. `ch32x035_pwr.c` *always* wants `debug.h`. Nasty.

And it only needs in one case where it needs a delay:

https://github.com/Community-PIO-CH32V/framework-wch-noneos-sdk/blob/342b5d0126878e9db048adeb91db256a2a1c5788/Peripheral/ch32x035/src/ch32x035_pwr.c#L143-L150

closed 2024-11-14 by maxgerhardt