Community-PIO-CH32V/platform-ch32v #76: baremetal-ch32v003 is not working after power off and resume.

RhonyD, 2024-11-14:
I build and flashed [‎baremetal-ch32v003]. Everything worked flawlessly, but as soon as I switched off the power and resumed everything stopped, no blink nothing. Please help.

Rockwell1799, 2024-11-14:
> I build and flashed [‎baremetal-ch32v003]. Everything worked flawlessly, but as soon as I switched off the power and resumed everything stopped, no blink nothing. Please help.

Can you give more info like a picture of the board you're using and post the code here!

maxgerhardt, 2024-11-14:
Not reproducable at all for me. I tested this with my `ch32v003f4p6_evt_r0` environment and WCH-LinkE, flashed the board

```
Checking size .pio\build\ch32v003f4p6_evt_r0\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.5% (used 256 bytes from 2048 bytes)
Flash: [          ]   3.1% (used 516 bytes from 16384 bytes)
Configuring upload protocol...
AVAILABLE: isp, minichlink, wch-link, wlink
CURRENT: upload_protocol = wch-link
Uploading .pio\build\ch32v003f4p6_evt_r0\firmware.elf
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2024-07-29-15:30)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

Warn : Transport "sdi" was already selected
Ready for Remote Connections
[wch_riscv.cpu.0] Target successfully examined.
** Programming Started **
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
===========================[SUCCESS] Took 3.08 seconds ===========================
```
Then unplugged the WCH-LinkE (to which the ch32v003 is connected to) and plugged it back in, the LED is still blinking!

RhonyD, 2024-11-15:
Thanks for you kind and prompt reply. Actually I was trying with a bare minimum board with a tssop adapter which is powered up using a AS1117 3v regulator. After your reply I took out the mini dev board which is a readymade development for CH32v003 and flashed the same code and it worked flawlessly. So the finding is that my bareminimum board is the culprit. I have redesigned a bareminimum board and now it's working perfectly, thanks again for your reply.

RhonyD, 2024-11-15:
if somehow the use of RAM can be redued then it would be great. In case of CHFUN framework the usage of RAM is very less.

RhonyD, 2024-11-15:
Another issue which I am facing is while trying to make my own lib file like the one which is given below:-
GPIO_CONFIG.h

#ifndef _GPIO_CONIFG_H
#define _GPIO_CONIFG_H
#include "ch32v00x.h"

#define u8      uint8_t

#define GPIO_PORT  GPIOD
#define GPIO_PIN    3
#define GPIO_PIN_1  2

void GPIO_INIT(uint8_t PIN);

#endif

and corresponding source file GPIO_CONFIG.c-

#include "GPIO_CONFIG.h"

void GPIO_INIT(u8 PIN)
{
    RCC->APB2PCENR&=~(RCC_APB2Periph_GPIOD);	
	GPIO_PORT->CFGLR &= ~(0xf<<(4*PIN));
	RCC->APB2PCENR|=(RCC_APB2Periph_GPIOD);	
	GPIO_PORT->CFGLR &= ~(0xf<<(4*PIN));
	GPIO_PORT->CFGLR |=(GPIO_CNF_OUT_PP|GPIO_Speed_50MHz)<<(4*PIN);

}

and then include #include "GPIO_CONFIG.h" in main.Now if I try building complier shows Error " "ch32v00x.h" No such file found.

what's the issue?

maxgerhardt, 2024-11-15:
[image]

Not reproducable at all. Please post the full project files.

RhonyD, 2024-11-16:
Actually I am putting the lib files in the lib folder. But you have put it in the source folder. May be that is the reason that in your case it worked. Thanks for your reply. I will get back to you soon.

maxgerhardt, 2024-11-16:
You need to adhere to the folder structure that PlatformIO wants. You can't just put .h and .c files directly into `lib/`, they need to be in a subfolder to be detected as library, such as `lib/gpio/GPIO_CONFIG{.c, .h}`. Note that also special rules apply to "libraries"; If they contain interrupt handlers, they must have a libary.json that says "libArchive":false, otherwise the `weak` interrupt handlers will still be used. Best to read up on https://docs.platformio.org/en/latest/librarymanager/index.html.

RhonyD, 2024-11-16:
Thanks for your reply and guidance. And most importantly all the workouts that you have suggested worked, Thanks again..

closed 2024-11-16 by maxgerhardt