Community-PIO-CH32V/platform-ch32v #77: Error: Failed to access data for an ELF segment???

Rockwell1799, 2024-11-14:
Hello
I have an issue with the following code:
```
#include <config.h>                               // user configurations
#include <system.h>                               // system functions
#include <gpio.h>                                 // GPIO functions
#include <debug_serial.h>                          // serial debug functions
volatile uint32_t count;
// ===================================================================================
// Main Function
// ===================================================================================
int main(void) {
  // Setup
  PIN_output(PIN_LED);                            // set LED pin to output

  // Loop
  while(1) {
    PIN_toggle(PIN_LED);                            // toggle LED
    DLY_ms(1000);                                  // wait a bit
    DEBUG_newline();
    DEBUG_printD(count++);                         // send string
  }
}
```
The code does compile, but I get the error: Failed to access data for an ELF segment.
When I remove the last 2 lines, the code uploads without any issue.
Does someone have an idea why I'm getting this error?
TIA

maxgerhardt, 2024-11-14:
Can you flash with `upload_protocol =wch-link` fine? What about `upload_protocol = minichlink`?

Rockwell1799, 2024-11-14:
I can't use minichlink,because I'm not using the winusb driver.

maxgerhardt, 2024-11-14:
Okay, then `upload_protocol = wch-link` should work if you use the WCH driver.

Can you upload the **exact** project files? It seems to be https://github.com/wagiminator/Development-Boards/blob/main/CH32X033F8P6_DevBoard/software/debug_serial but with some modifications.

Rockwell1799, 2024-11-14:
With wch-link, it uploads the code, but the LED is not blinking anymore!
[blink.zip](https://github.com/user-attachments/files/17758476/blink.zip)
I've attached the project files as zip below.

maxgerhardt, 2024-11-14:
Look in the debugger. You're hanging up in the DEBUG_write() function which waits for the character to be written, but that can never happen because your `main()` function never initializes the UART by calling into `DEBUG_init()`.

[image]

maxgerhardt, 2024-11-14:
Thanks to your project, I can reproduce the error
```
22:30:04 [INFO] Connected to WCH-Link v2.14(v34) (WCH-LinkE-CH32V305)
22:30:04 [INFO] Attached chip: CH32X035 [CH32X035F8U6] (ChipID: 0x035e0601)
22:30:04 [INFO] Chip ESIG: FlashSize(62KB) UID(cd-ab-28-c5-49-bc-3e-2d)
22:30:04 Error: Failed to access data for an ELF segment.
[INFO] Flash protected: false
22:30:04 [INFO] Read .pio\build\CH32X035\firmware.elf as ELF format
*** [upload] Error 1
```
and will look into it. Maybe we need a `wlink.exe` update, or a bug should be opened there.

maxgerhardt, 2024-11-14:
This error is reproducable in the latest version of wlink and now tracked in https://github.com/ch32-rs/wlink/issues/77.

Please use `upload_protocol = wch-link` since it works fine.

You can also debug with it. Just open the debug sidepanel and click on the green Play button next to "PIO Debug". You can even view all perpiheral registers.

[image]

closed 2024-11-14 by maxgerhardt

maxgerhardt, 2024-11-15:
Will update the PlatformIO binaries for `wlink` once https://github.com/ch32-rs/wlink/pull/79 is released.

maxgerhardt, 2024-11-16:
The updated binaries have been submitted to registry as `tool-wlink` and are awaiting approval by the PlatformIO team.

>The package has been accepted for moderation. We will notify you on email when process it.

maxgerhardt, 2024-11-16:
All binaries were actually updated in https://github.com/Community-PIO-CH32V/tool-wlink/ (PIO registry unused for that package), so it should all work now.