Community-PIO-CH32V/platform-ch32v #78: strcat function is not working with  framework = ch32v003fun.

RhonyD, 2024-11-28:
strcat function is not working with  framework = ch32v003fun in VS code.
message from compiler
/.platformio/packages/toolchain-riscv/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld: /tmp/ccXXAkkp.ltrans0.ltrans.o: in function `.L244':
<artificial>:(.text.startup.main+0x378): undefined reference to `strcat'
what to do please suggest.

maxgerhardt, 2024-11-28:
The ch32v003fun framework is all about being as small as possible. It deliberately **disables** the inclusion of the compiler-provided standard C library (by using the `-nostdlib` compiler and linker flag) in order to be able to implement all standard C functions that it needs itself. See e.g.

https://github.com/cnlohr/ch32v003fun/blob/1ea96b14248628a209816963821b088dc8ecf506/ch32v003fun/ch32v003fun.c#L216-L221

`strcat()` is not implemented there. So, this is not strictly a bug. If you think it should be provided there, file a ticket in https://github.com/cnlohr/ch32v003fun. 

You can also create the function yourself, as illustrated by e.g. [the musl C library](https://git.musl-libc.org/cgit/musl/tree/src/string/strcat.c)
```cpp
char *strcat(char *restrict dest, const char *restrict src)
{
	strcpy(dest + strlen(dest), src);
	return dest;
}
```

Of course, you can also deviate from the clean, holy path and re-enable the inclusion of the compiler-provided C library, by adding
```ini
build_unflags = -nostdlib
```
into the `platformio.ini`.

RhonyD, 2024-11-29:
Thanks for your reply. I have implemented my own strcat function and it worked nicely. And there is no point of deviating from the clean, holy path of ch32v003fun framework  and I am developing all my libraries based upon the same framework. it was just out of curiosity but anyways thanks again for your help and reply.

closed 2024-11-29 by maxgerhardt