Community-PIO-CH32V/platform-ch32v #79: enable and disable code protect using minichlink

RhonyD, 2024-11-30:
how to enable and disable code protect using minichlink?

maxgerhardt, 2024-11-30:
According to https://github.com/cnlohr/ch32v003fun/blob/bf73d9ad18b525059e5b80b1052a3c4dccc3463b/minichlink/minichlink.c#L733-L734 with the `-P` flag, so you could say in the `platformio.ini`
```ini
upload_flags = -P
```
(or add it onto maybe other existing `upload_flags`).

You may not be able to flash again after that, until read protection is disabled again, not sure.  `-p` will unprotect the flash again.

RhonyD, 2024-12-01:
I can do it using the mourn river ide and then going to the configure uploader section, But it will be convenient to do so from within the vscode environment. Please see if it is possible.

RhonyD, 2024-12-01:
in minichlink when given ./minichlink -p following are the outputs

[image]
[image]
 this shows that enable and disable code protect  can be done using minichlink but in case of vscode it's not possible yet.

maxgerhardt, 2024-12-01:
Fixed in https://github.com/Community-PIO-CH32V/platform-ch32v/commit/9078ce4b9211753d7cb088ec0bf3cc351515d0a5

[image]

closed 2024-12-01 by maxgerhardt

RhonyD, 2024-12-03:
Thanks Bro, it worked like a charm, GREAT JOB.... KUDOS TO YOU....