Community-PIO-CH32V/platform-ch32v #83: funconfig.h not found when in include folder where it belongs

markusdd, 2025-02-09:
I noticed that when using ch32v003fun framework includes are not found when they are in the proper include directory, but rather they have to be put directly next to the main sources inside src.

Is this intentional or an oversight in include path settings in the project template?

maxgerhardt, 2025-02-09:
Do you mean that `funconfig.h` is not found when it's placed inside `include/` rather than `src/`? 

Or do you mean that `#include <ch32v003fun.h>` fails when its written inside a file in `include/`?

maxgerhardt, 2025-02-09:
While I did fix the first point in 349fa5a8fe58a8b755dcdb88d408d11f14127d3e, the second point should already work irregardless of that.

[image]

(To update the platform, [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) -> `pio pkg update -g -p ch32v`)

maxgerhardt, 2025-02-09:
Right, so going by the title it's rather the first one, which is now fixed per commit above.

[image]

closed 2025-02-09 by maxgerhardt

markusdd, 2025-02-09:
Can confirm it now works automatically and properly after the update.
My install was so new I assumed there weren't any but I was wrong haha.

Thx for the quick response!