Community-PIO-CH32V/platform-ch32v #88: No longer working on windows with space in path

Zeron, 2025-04-09:
73155b013000312f634907ba62c86dc4e6659 breaks compile on windows if there is a space in the path

`Building C:\Users\Dan\Documents\PlatformIO\Projects\Pin Test\.pio\build\WeAct_CH32V203C8T6/PinTest_WeAct_CH32V203C8T6.lst
riscv-none-embed-objdump: 'Test\.pio\build\WeAct_CH32V203C8T6/PinTest_WeAct_CH32V203C8T6.elf': No such file
riscv-none-embed-objdump: 'Test\.pio\build\WeAct_CH32V203C8T6/PinTest_WeAct_CH32V203C8T6.lst': No such file
*** [.pio\build\WeAct_CH32V203C8T6\PinTest_WeAct_CH32V203C8T6.elf] Error 1`

maxgerhardt, 2025-04-09:
Sorry, fixed in 2a620b07562894957e59228a99ce1c837ca08172 again.

Please update with `pio pkg update -g -p ch32v` on the [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli).

Zeron, 2025-04-09:
Working now - Thanks for the quick response.

closed 2025-04-09 by Zeron