Community-PIO-CH32V/platform-ch32v #89: Cannot compile for several boards using GIT version

Zeron, 2025-04-09:
With the following platform.ini I get build failures for most genericXXX boards

`[env]
platform = https://github.com/Community-PIO-CH32V/platform-ch32v.git#2a620b07562894957e59228a99ce1c837ca08172
platform_packages = framework-arduino-openwch-ch32@https://github.com/openwch/arduino_core_ch32.git
framework = arduino
board_build.core = openwch
upload_protocol = wch-link

[env:genericCH32L103C8T6]
board = genericCH32L103C8T6

[env:genericCH32V003A4M6]
board = genericCH32V003A4M6

[env:genericCH32V003F4P6]
board = genericCH32V003F4P6

[env:genericCH32V003F4U6]
board = genericCH32V003F4U6

[env:genericCH32V003J4M6]
board = genericCH32V003J4M6

[env:genericCH32V103R8T6]
board = genericCH32V103R8T6

[env:genericCH32V203C6T6]
board = genericCH32V203C6T6

[env:genericCH32V203C8T6]
board = genericCH32V203C8T6

[env:genericCH32V203C8U6]
board = genericCH32V203C8U6

[env:genericCH32V203G6U6]
board = genericCH32V203G6U6

[env:genericCH32V203G8R6]
board = genericCH32V203G8R6

[env:genericCH32V203RBT6]
board = genericCH32V203RBT6

[env:genericCH32V307RCT6]
board = genericCH32V307RCT6

[env:genericCH32V307VCT6]
board = genericCH32V307VCT6

[env:genericCH32V307WCU6]
board = genericCH32V307WCU6

[env:genericCH32X035G8U6]
board = genericCH32X035G8U6
`

If I remove platform_packages = framework-arduino-openwch-ch32@https://github.com/openwch/arduino_core_ch32.git many more compile.

|Part|GIT|1.0.4|
|--------|--------|--------|
|genericCH32L103C8T6|Works|Works|
|genericCH32V003A4M6|Invalid board option 'build.arduino.openwch.variant_h'|Invalid board option 'build.arduino.openwch.variant_h'|
|genericCH32V003F4P6|Works|Works|
|genericCH32V003F4U6|Works|Works|
|genericCH32V003J4M6|Invalid board option 'build.arduino.openwch.variant_h'|Invalid board option 'build.arduino.openwch.variant_h'|
|genericCH32V103R8T6|fatal error: variant_CH32V103R8T6.h: No such file or directory|Works|
|genericCH32V203C6T6|fatal error: variant_CH32V203C6.h: No such file or directory|Works|
|genericCH32V203C8T6|fatal error: variant_CH32V203C8.h: No such file or directory|Works|
|genericCH32V203C8U6|fatal error: variant_CH32V203C8.h: No such file or directory|Works|
|genericCH32V203G6U6|fatal error: variant_CH32V203G6.h: No such file or directory|Works|
|genericCH32V203G8R6|fatal error: variant_CH32V203G8.h: No such file or directory|Works|
|genericCH32V203RBT6|fatal error: variant_CH32V203RB.h: No such file or directory|Works|
|genericCH32V307RCT6|Invalid board option 'build.arduino.openwch.variant_h'|Invalid board option 'build.arduino.openwch.variant_h'|
|genericCH32V307VCT6|Works|Works|
|genericCH32V307WCU6|Invalid board option 'build.arduino.openwch.variant_h'|Invalid board option 'build.arduino.openwch.variant_h'|

Running on Windows 11 64-Bit

Zeron, 2025-04-09:
I can compile for CH32V103, CH32V203, in the Arduino IDE using the latest GIT.  Only fails using Platformio.

maxgerhardt, 2025-04-09:
You cannot fully use
```
platform_packages = framework-arduino-openwch-ch32@https://github.com/openwch/arduino_core_ch32.git
```
because many parts of the PlatformIO builder script are not merged into that repository yet. This depends on  https://github.com/openwch/arduino_core_ch32/pull/147.

Canonically, for CH32V00x, only the CH32V003F4 variant is defined, not the others: https://github.com/Community-PIO-CH32V/arduino_core_ch32/tree/main/variants/CH32V00x

I will take a look at them again though.

Zeron, 2025-04-09:
> You cannot fully use
> 
> platform_packages = framework-arduino-openwch-ch32@https://github.com/openwch/arduino_core_ch32.git
> 
> because many parts of the PlatformIO builder script are not merged into that repository yet. This depends on [openwch/arduino_core_ch32#147](https://github.com/openwch/arduino_core_ch32/pull/147).
> 

I switched to
platform_packages = framework-arduino-openwch-ch32@https://github.com/Community-PIO-CH32V/arduino_core_ch32.git
an now I only get the four Invalid board option 'build.arduino.openwch.variant_h' errors.

> Canonically, for CH32V00x, only the CH32V003F4 variant is defined, not the others: https://github.com/Community-PIO-CH32V/arduino_core_ch32/tree/main/variants/CH32V00x

I just grabbed all the board that have arduino framework listed in https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/boards