Community-PIO-CH32V/platform-ch32v #90: board directive "build.stack_size" ignored

roglio, 2025-04-30:
I for a project of mine [RSVP_TinyBasic](https://github.com/arduinousergroupcagliari/RVPC_TinyBasic), I was forced to resize the stack, but the foreseen parameter "build.stack_size" is ignored and also not defined in boards .json definitions.

Being this MCU very restricted in resource, I hope will be foreseen to enable this parameter. At moment I'm forced to specify to modify the noneos_sdk.py to manually modify the stack_size (line 44).

Thanks.
L.

roglio, 2025-04-30:
Sorry, I was not aware of the correct way to define this parameter: board_build.stack_size works.

closed 2025-04-30 by roglio

maxgerhardt, 2025-04-30:
Correct, per general PlatformIO documentation, if you want to influce a board parameter like `build.stack_size`, you have to write it in the `platformio.ini` with a `board_` prefix:

https://docs.platformio.org/en/latest/boards/atmelavr/uno.html

This is a documentation issue, that needs to be solved as a whole. Will be tracked in #12.