Community-PIO-CH32V/platform-ch32v #97: platform support CH32V002, debug

Igor26974, 2025-06-26:
Is it possible to add debug for CH32V002 to platformio?

maxgerhardt, 2025-06-29:
The board definitions for CH32V002xx (and others) have been added in 18be44e925dc6b3b4fb67285922451e72efbdb26. But I guess this needs an OpenOCD update too. What exact ch32v002 chip do you have?

Igor26974, 2025-06-29:
Blink worked!
**!!!!!!! THANK YOU VERY MUCH !!!!!!!**
and debug works!
I'll check ADC now

maxgerhardt, 2025-06-29:
Great! You should be able to import any of the ADC examples from https://github.com/openwch/ch32v002_004_005_006_007/tree/main/EVT/EXAM/ADC.

A general note: When importing the SDK examples, you
* **don't need to** copy the `ch32v00X_conf.h` (already included in SDK package)
* **don't need to** copy the `system_ch32v00X.{.c, .h}` (already included in SDK package and configurable via `build_flags`, default is 48MHz from HSI)
* if you want to see printf output via UART, make sure you connect the right TX pin to your WCH-LinkE's RX pin: default is TX = PD5. (https://github.com/openwch/ch32v002_004_005_006_007/blob/main/EVT/EXAM/SRC/Debug/debug.h#L24-L47). I updated [examples/uart-printf-none-os](https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/examples/uart-printf-none-os) to make this clearer.
* if you want to see printf output SDI (debug connection), you need to set `build_flags = -DSDI_PRINT=1`  and use [the project task](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks) Platform -> "Enable SDI printf (NoneSDK)" to make your WCH-LinkE's virtual COM port spit out the SDI prints.

Igor26974, 2025-06-29:
For some reason I can't start a constant ADC measurement via DMA in a cyclic measurement of the A4 channel on platformio.
Does the CH32v002 have an ADC auto calibration mode?
Do you have an example by any chance?

maxgerhardt, 2025-06-29:
The wrong SRAM sizes were fixed in the latest update. (You need to update in the same way as before).

I'm not sure about ADC "auto calibration mode". I only find a "tCAL Calibration time" (100 ADC clock cycles) in the datasheet. I guess that's something the ADC does automatically when you turn it on? The V00X SDK does not have the `ADC_StartCalibration()` function like other chips do.

Or do you mean you want to measure V_Ref with the ADC, which is the "ADC_IN8" channel?

maxgerhardt, 2025-06-30:
Ah, actually I've just remembered that we don't use the PlatformIO Registry for the RISC-V toolchain. I just updated the GCC12 toolchain and the build is passing for all frameworks and examples.

So, if you  `pio pkg update -g -p ch32v` again, your project should now compile with GCC12 **and** the correct multiply instruction for ch32v00x (x != 3). 

In the ADC simple example, the code line

```cpp
        val_mv = (ADC_val * 3300 / 4096); // Assuming VCC = 3300mV, not really calibrated
```
compiles to a `mul` instruction now instead of `__muli4`. However, division and modulo have no hardware support, so it will still call a function there.

```
     558:	028606b3          	mul	a3,a2,s0

Igor26974, 2025-07-04:
Thank you for your work! 
I've got it all worked out.

I still need to store some bytes(int8_t, uint16_t) in EEPROM. 
And how can I use flash ch32v002 to store user bytes.
Is there an example for reading and writing?

maxgerhardt, 2025-07-04:
Great!

They have a flash erase/write/read example at https://github.com/openwch/ch32v002_004_005_006_007/blob/main/EVT/EXAM/FLASH/FLASH_Program/User/main.c. It uses two very easy convience function for erasing and writing. Reading is done with just a `mempcy()` from the flash address.

https://github.com/openwch/ch32v002_004_005_006_007/blob/b5768ab7f6bb5fe824d83ecc31e4b637251571e9/EVT/EXAM/SRC/Peripheral/inc/ch32v00X_flash.h#L140-L141

So all you have to do is decide in which flash page(s) you want to store user data, usually taken from the last few pages of flash.

[image]

The CH32V002 also has a "256 byte user-defined information store for user option bytes storage", so if your data fits in there, you can use that space. But that's also just one page of flash.

[image]

Igor26974, 2025-07-04:
I'm fine with up to 40 bytes.
I use user option bytes storage but only 2 bytes "data0" and "data1" 
And with the rest of user option bytes storage I have not figured out which addresses can be used.

maxgerhardt, 2025-07-06:
In any case, since the original issue of general CH32V002 support is resolved, I will close this issue. For seperate issues with the platform code, open a new issue.

closed 2025-07-06 by maxgerhardt