Community-PIO-CH32V/platform-ch32v #98: The iic issue in ch32v003

ztf402, 2025-06-28:
When I trying to use iic in ch32v003, the arduino library may cost a lot of memory to initialize the iic (about 7000bytes),which is almost the half of the tatal ROM.Frankly speaking ,this way can success initalizing.When I trying to cut down the rom usage by replace the arduino framework with None-os,but it can't work correctly.Neither the offical (WCH.)'s example code nor my custom code did not work in it.But,when I paste them to the MonuRiver Studio,it can work normally.

ztf402, 2025-06-28:
I have compared the complied file between MonuRiver's and platform's,both have build flag -Os,but the output files' size is different.I think it might be a bug in the build flag in platformio's ch32v

maxgerhardt, 2025-06-28:
This is really not enough to be reproducable. Can you upload your PlatformIO project to Github as a whole?

Also what version of MounRiver are you using? Right now the platform has not yet been updated to use the newer RISCV GCC compiler or the newer NoneSDK version, so size differences are expected. This will be updated soon.