Community-PIO-CH32V/platform-ch32v #100: PlatformIO Build error on macOS Intel, Bad CPU type in executable

donly, 2025-07-01:
It works very well on Apple Silicon. 

And I found the issue maybe caused by the toolchain-riscv.

```
otool -hv ~/.platformio/packages/toolchain-riscv/bin/riscv-wch-elf-g++
/Users/donly/.platformio/packages/toolchain-riscv/bin/riscv-wch-elf-g++:
Mach header
      magic  cputype cpusubtype  caps    filetype ncmds sizeofcmds      flags
MH_MAGIC_64    ARM64        ALL  0x00     EXECUTE    20       2000   NOUNDEFS DYLDLINK TWOLEVEL WEAK_DEFINES
 BINDS_TO_WEAK PIE
```

donly, 2025-07-01:
I also found that it can be built under a specific version of toolchain-riscv using @Zeron's method #89

maxgerhardt, 2025-07-01:
True. The repo for the Mac toolchain, https://github.com/Community-PIO-CH32V/toolchain-riscv-mac/ should only have Intel x86 binaries and a new repo should be made and used for Mac ARM64. This will be corrected shortly.

maxgerhardt, 2025-07-02:
>Hello Maximilian,
>Sorry for the inconvenience. Currently, we only support ARM architecture, and there are no plans to support Intel architecture in the near future.

>Regards,
>MounRiver Studio

Well, they said "no". Too bad for Intel Mac users.

I'm now trying to obtain the GCC12 source code from them. They must do this due to the GPLv3 license of GCC. But, companies rarely comply..

maxgerhardt, 2025-07-04:
I asked them to correct their website description "MacOS_x64 -> MacOS_ARM64" and asked them for source code and they said

>Hello,
>Thank you for your suggestions and help. We will revise the description on the website in the upcoming updates.
>However, we apologize that due to our toolchain being in a phase of rapid iteration, we are unable to provide the relevant source code at the moment.
>We will consider adding support for the x86_64 toolchain in the future.
>
>Regards,
>MounRiver Studio	
>2025.7.3

Aww too bad! They are in phase of rapid iteration, so sadly they can't abide by what the licence says they must do! I must give the poor company a break, they can't just give out the source code as they're required to by the license.