Community-PIO-CH32V/platform-ch32v #101: Arduino Framework USB CDC/TinyUSB library include error

Mapy542, 2025-07-06:
I apologize, this may not be an issue at all with your platform, but I have been unable to figure out how to use the USB CDC as a serial in the Arduino framework. I have tried to use both the openwch and ch32v build cores to no success. I have not been able to generate a COM device from the USB ports of a generic dev board. I can use the port to upload programs, though.

Best I can assume, this should work as a main.cpp file:
`
#include <Arduino.h>
#include <stdio.h>

void setup() { Serial.begin(9600); }

void loop() {
    Serial.println("Hello, World!");
    delay(1000);
}
`
ini File:
`
[env:genericCH32V307VCT6]
platform = ch32v
board = genericCH32V307VCT6
framework = arduino
upload_protocol = isp
board_build.core = openwch
board_upload.maximum_size = 294912
board_upload.maximum_ram_size = 32768
`

When I try openwch or ch32v build cores, nothing happens. When I add: 
`
build_flags = 
	-D USE_TINYUSB
`
to openwch, I get 
`
 8 | #include "Adafruit_USBD_CDC.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from C:\Users\elimb\.platformio\packages\framework-arduino-openwch-ch32\cores\arduino\wiring.h:48,
                 from C:\Users\elimb\.platformio\packages\framework-arduino-openwch-ch32\cores\arduino\Arduino.h:36,
                 from C:\Users\elimb\.platformio\packages\framework-arduino-openwch-ch32\cores\arduino\HardwareSerial.cpp:27:
C:\Users\elimb\.platformio\packages\framework-arduino-openwch-ch32\cores\arduino\WSerial.h:8:10: fatal error: Adafruit_USBD_CDC.h: No such file or directory
`

I can see the TinyUSB library has been git sub-module included into the libraries folder of the framework. However, it doesn't seem to be found by the rest of the PIO platform.
Any help on getting USB communication set up via any means is appreciated. I do need to stick to C++, and ideally the Arduino framework.

maxgerhardt, 2025-07-09:
There is indeed a fault in maybe the PlatformIO builder script for the Arduino core. The clock source is not set correctly. With no further configuration, the chip stays at its HSI (8 MHz) configuration, which the USB startup code detects and *rejects*, as it needs at least 48 MHz from e.g. HSE or HSI (better HSE). 

Now with the addition of `-DSYSCLK_FREQ_144MHz_HSE=144000000` in the `build_flags` of the tinyusb-arduino example to nudge the `system_ch32v30x.c` to initialize the clock properly, I **do** get a USB device on one of the USB ports (the dev board has two!) called "TinyUSB Serial (Interface 0)". 

Then, Windows driver weirdness kicks in and it loads a driver from WCH that's meant for the WCH-Link(E) instead of the serial port drivers.

[image]

So using Zadig, I have to revert that and load USBCDC drivers, which then gives me the actual COM port. 

[image]

Opening that serial port with either the PlatformIO serial monitor ([project tasks](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks) -> Monitor) or HTerm, there is data coming in, **given** that the virtual DTR line is asserted. The TinyUSB serial port will otherwise send no serial data. However, there seems to a bug with the buffer, in the sense that if the serial monitor is not opened immediately after the device boots, the device never sends data. So I had to press the reset button on the board and quickly reconnect the serial monitor, only then it would send data.

[image]

So we have at least two distinct bugs to work on here.

maxgerhardt, 2025-07-14:
Very interesting, maybe the buffer bug was just a randomness on my side. The clock configuration was restructued in https://github.com/Community-PIO-CH32V/platform-ch32v/commit/58b22212830cd2d4f7bd6f1a9ee4203ad7cd2074, and is documented in https://pio-ch32v.readthedocs.io/en/latest/clock%20source%20configuration.html. As thus, there is no need for the `-DSYSCLK_FREQ_144MHz_HSE=144000000` anymore, just setup the clock source to hse+pll. The default frequency is already 144MHz.

The GPIO pins PA15 and PB4 are by default **not** accessible in the OpenWCH core's "digital pin" method, because it only deals with the "Arduino header" on a specific CH32V307 dev board. There is only digital pins 0 (D0) down to D21, as in 

https://github.com/openwch/arduino_core_ch32/blob/d76716239cdf8a084a5045c3dfd3151b3f69eeec/variants/CH32V30x/CH32V307VCT6/variant_CH32V307VCT6.cpp#L20-L44

You can still write code to blink PA15 and PB4, but you'll have to use a slightly different method:

```cpp
#include <Arduino.h>

/* PA_15 **not** PA15!!! Needs to use "fast" API that works on PinName type.*/
#define LED PA_15

void setup() {
  /* there is no pinModeFast() that natively takes PinName arg, so we have to call the underlying API */
  pin_function(LED, CH_PIN_DATA(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_PP, 0, 0));
}

void loop() {
  digitalWriteFast(LED, HIGH);
  delay(1000);
  digitalWriteFast(LED, LOW);
  delay(1000);
}
```
please let me know if you get to blink PA15 and PB4 with that.

Mapy542, 2025-07-16:
Thank you for the help with pin-interfacing. I figured it was something like that. Between now and my last reply, I checked all the pins and made a list of which ones work in the default digitalWrite. I figured it was a software setup. It's super helpful to be able to interface with all of them. Both A_15 and B_4 work, plus a few others I tested.

I checked, and the serial works well too, but only in hse+pll. I see there are other options, but only checked against hsi+pll. The internal version did not work for me. However, the default assignment and ini modification to hse both report the correct values during compilation 
[image]
here.

Mapy542, 2025-07-21:
Ok, that makes a ton of sense.
I tried both HS and FS, neither worked.

That seems reasonable. I just wasn't aware that the internal clock was not stable enough. Maybe emit a warning during compilation if hsi is selected?