Community-PIO-CH32V/platform-ch32v #107: How to fully remove platform-ch32v

tonyjonespdx, 2025-08-30:
For an unknown reason when running visual-studio-code the platform-io extension reported "This extension has been disabled because it does not support virtual workspaces".

**I could not figure out the issue** so I removed **everything** using uninstallers (visual-studio-code,  all extensions).  Then removed all files from filesystem.

I then reinstalled visual-code:

C:\Windows\System32>code --list-extensions
ms-vscode.cpptools

If I try to reinstall platform-io it still reports "This extension has been disabled because it does not support virtual workspaces"

More interestingly in the explorer pane there is still "platform-ch32v" listed.

If I try to click on it I get an alert "Unable to resolve resource vscode-vfs://github/Community-PIO-CH32V/platform-ch32v".

This is obviously a "virtual workspace" which is maybe related to the "does not support virtual workspaces" error.

I have no "\Users\<user>\.platformio" directory as I removed it when I uninstalled.  

How do I clean this mess up?

maxgerhardt, 2025-08-30:
`C:\Users\<user>\.platformio` is where the PlatformIO core is installed, and all installed platforms. So it should have killed it when you removed that. Of course the PlatformIO VSCode extension is installed somewhere else, but those are VSCode internals.

It would be very weird if the extension just broke, I'm running VSCOde 1.103.2 and PlatformIO works just fine. 

[image]

What VSCode version are you running?

tonyjonespdx, 2025-08-30:
I am running the same version (latest) of vscode 1.103.2.  As I said I removed it, wiped all directories and reinstalled it.

I have PLATFORMIO_CORE_DIR=Z:\VisualStudioCode\Extensions\PlatformIO

There is nothing in this directory (because I don't have platformio presently installed) as confirmed by "code --list-extensions".

I have no C:\Users\<user>\.platformio directory, because I'm specifying PLATFORMIO_CORE_DIR.

Yet still I see "Platform-CH32V" present.

[image]

In fact I can't install Platform IO.   If I try to it complains "This extension has been disabled because it does not support virtual workspaces" which I suspect is related to the

maxgerhardt, 2025-08-30:
This looks to me like you just have a stale project open. Just press "OK" and open a new VSCode window with File -> new window, then close the old one.