Community-PIO-CH32V/platform-ch32v #108: Support for custom startup file

dzanis, 2025-09-01:
Hello,

I needed to open and edit a project created in **MounRiver Studio 2 (MRS2)** inside **VSCode** (because MRS2 does not support version control).
For this I installed **platform-ch32v**, and used the following `platformio.ini`:

```ini
[platformio]
src_dir = User

[env:CH32X033F8P6]
platform = ch32v
board = genericCH32X033F8P6
framework = noneos-sdk
board_build.ldscript = $PROJECT_DIR/Ld/Link.ld
board_build.use_lto = yes
board_build.use_builtin_startup_file = no
board_build.use_builtin_system_code = no
board_build.use_builtin_debug_code = yes
board_build.cpp_support = no

build_flags =
  -I User

monitor_speed = 115200
```

The problem is that the **startup file (`Startup/startup_ch32x035.S`) is not included in the build**.
It only works if I copy `startup_ch32x035.S` into the `User` folder, but that breaks the original project structure from MRS2.

I could not find a way to include the startup file directly from the `Startup` directory, until I added this code in
`.platformio/platforms/ch32v/builder/frameworks/noneos_sdk.py`  [there](https://github.com/Community-PIO-CH32V/platform-ch32v/blob/7a00b2c74263878bece4c3e1ad7716783a47b295/builder/frameworks/noneos_sdk.py#L121): 

```python
startup_filename = board.get("build.startup", None)

if get_flag_value("use_builtin_startup_file", True) or startup_filename:
    env.Append(CPPPATH=[join(FRAMEWORK_DIR, "Startup")])
    if not startup_filename:
        startup_filename = get_startup_filename(board)
    startup_file_filter = "-<*> +<%s>" % startup_filename
    env.BuildSources(
        join("$BUILD_DIR", "FrameworkNoneOSStartup"),
        join(FRAMEWORK_DIR, "Startup"),
        startup_file_filter
    )
```

With this patch, it is possible to specify the startup file path in `platformio.ini`:

```ini
board_build.startup = $PROJECT_DIR/Startup/startup_ch32x035.S
```

But this local patch is temporary.Could you please add this feature to the project?
It would make it much easier to keep MRS2 projects compatible with PlatformIO without breaking their folder structure.

Thanks!

maxgerhardt, 2025-09-02:
Added in a slightly more sophisticated way (automatically inferring `$PROJECT_DIR` relativity if path is not absolute) via 384e1129bd89de4843b28943a3bd526e3ab47a3e and documented via https://github.com/Community-PIO-CH32V/docs/commit/c39be149d9f175b6dcdf53d3435e34e6f34bef50 in https://pio-ch32v.readthedocs.io/en/latest/none%20os%20sdk.html#startup-file

closed 2025-09-02 by maxgerhardt

dzanis, 2025-09-02:
> Added in a slightly more sophisticated way (automatically inferring `$PROJECT_DIR` relativity if path is not absolute) via [384e112](https://github.com/Community-PIO-CH32V/platform-ch32v/commit/384e1129bd89de4843b28943a3bd526e3ab47a3e) and documented via [Community-PIO-CH32V/docs@c39be14](https://github.com/Community-PIO-CH32V/docs/commit/c39be149d9f175b6dcdf53d3435e34e6f34bef50) in [official docs](https://pio-ch32v.readthedocs.io/en/latest/none%20os%20sdk.html#startup-file).

Thanks a lot for adding this feature!
It's great that information was added to the output and that there's now a check for an incorrect file path.

One small detail: when using
```ini
board_build.startup = $PROJECT_DIR/Startup/startup.S
````

or

```ini
board_build.startup = Startup/startup.S
```

After assembly, the object file (startup.S.o) is created directly in the project folder.
For example, in my case it appeared directly in the project folder with the following subfolders:
`OneDrive\Documents\CH32X033\CH32X033_runinram_test\Startup\startup_ch32x035.o`
 Is created in the project root instead of under **.pio/build/....**

This is of course not critical, but the aesthetics and project structure suffer a bit compared to the builtin startup files (which are placed under **$BUILD_DIR**).